calyxir/calyx #331: Make primitives use invokable go-done interface

rachitnigam, 2020-12-28:
The invokable go-done interface states that the outputs from an invoked component stay available till the next invocation. Currently, out primitive library modules don't follow this convention. Instead, they usually reset their output after `done == 0`. 
Change this and make them invokable.

Once primitives are invokable, they also become `share`able.

- [ ] `std_mult_pipe`
- [ ] `std_mod_pipe` and `std_div_pipe` (after #330)
- [ ] `std_sqrt`

cgyurgyik, 2021-01-24:
For my understanding, how exactly is the `mult_pipe` made invokable without creating a latch? I was thinking an extra boolean value in the if statement that checks to see the operation has been run once, but I'm not entirely convinced that's correct.

rachitnigam, 2021-01-24:
As far as I know, there isn't a way to do this without adding a latch.

closed 2021-10-06 by rachitnigam