calyxir/calyx #338: Represent bounded loops in Calyx

rachitnigam, 2021-01-03:
During `static-timing` compilation, all loops are assumed to be unbounded and given no `"static"` attribute. This means that control sub-programs with `while` and components with `while` cannot be statically compiled since the requirement of static compilation is all children be statically compilable.

However, Dahlia has `for` loops which are always bounded. Similarly, there might be other backends that want to represent bounded loops and take advantage of static compilation. We have two options:

1. Introduce a new `repeat` control operator that desugars to plain `while` loops (making it a so-called "level-2" operator). The repeat operator would run the control sub-program a finite number of times and can be statically compiled.
2. Add attributes to the control language (extension of #311) and give `while` loops an attribute (`"repeats"`) which specifies the number of iterations the loop is expected to take.

rachitnigam, 2021-01-04:
How about:
```
@bound(12) while le.out with cond0 { ... }
```

Where the `@bound(12)` says that the loop runs 12 times.

sampsyo, 2021-01-04:
Sounds excellent to me! I like the attribute here a bit more than adding a whole new control construct because it keeps things a bit more orthogonal (bounded & unbounded can get compiled the same way). And a trip-count annotation does seem like the way to go, specifically.

rachitnigam, 2021-03-29:
@ai-cu are you working on this? If so, please assign yourself the issue and change the "status" label to "in-progress".

PR #534 (2021-06-14): Infer static timing for bounded loops.
Closes #338.
closed 2021-06-14 by cgyurgyik