calyxir/calyx #343: Externalize should use cell attributes

rachitnigam, 2021-01-04:
Instead of making fake cells like `std_mem_d1_ext`, we should define an `@external(1)` attribute for cell declarations and change the `externalize` pass to use them:
```
cells {
  @external(1) A = prim std_mem_d1(...);
}
```

A benefit of doing this is that *any* cell can be externalized, not just cells with fake `_ext` counterparts.

TODO:
- [ ] Update the Dahlia frontend to use the attribute
- [ ] Change the `externalize` pass
- [ ] Remove `_ext` primitive cells.

sampsyo, 2021-01-04:
Great! Making this generic, for any old component, seems like a clean win---although I can't at the moment think of what else we would want to externalize? Maybe off-chip memories or host interfaces?

rachitnigam, 2021-01-05:
I'm imagining anything that you'd want to debug could also be externalized in some manner: externalize the `sqrt` module and drive it using Verilator or something.

sampsyo, 2021-01-05:
Sure, seems interesting!

closed 2021-01-25 by sgpthomas