calyxir/calyx #358: NTT Pipeline in FuTIL

cgyurgyik, 2021-01-22:
## Algorithm
This is a FuTIL-generated pipeline for  the Cooley-Tukey algorithm that avoids the bit-reversal step in the on-the-fly calculation by using `phis` in bit reversed order. This algorithm (below) is presented in [Speeding up the Number Theoretic Transform for Faster Ideal Lattice-Based Cryptography](https://www.microsoft.com/en-us/research/wp-content/uploads/2016/05/RLWE-1.pdf).

[image]

## Preconditions
- N = length of `a` is a power of 2.
- `q` meets the following criteria:
(1) q = 2 * n * k + 1, for some k.
(2) q is a prime number.
(3) q >= min_modulus > n
(4) q is equivalent to 1 mod 2N.
- `phis` are powers of `phi` in bit-reversed order

## Hardware Used

2 * `std_mem_d1_ext` (`a`, `phis`)

4 * N `std_reg` for parallelization.
- `A` registers are used to save the output of calculations.
- `r` registers are used within the calculation.
- `mul` registers are used to save the output of the multiply with `phi`.
- `phi` registers are used to save the value of `phi` for each index.

N * `std_smod_pipe`
N * `std_smult_pipe`
N * `std_sadd`
N * `std_ssub`

Currently, the modulus value `q` is just hard-wired to be a constant, e.g. if `q = 97`, then I just do `mod_pipe.right = 32d’97`, rather than using N * `std_const` as well. Perhaps there's some negative implications to this that I'm missing?

## Control
There are several stages in the control flow on top of the pipeline stages one would expect in NTT. For N=4, it would look something like this:
```
control {
  seq {
    // Write from memory to registers.
    seq { preamble_0; preamble_1; preamble_2; preamble_3; }

    par {
      seq { stage0_row0_mul; stage0_row0_mod; }
      seq { stage0_row1_mul; stage0_row1_mod; }
      seq { stage0_row2_mul; stage0_row2_mod; }
      seq { stage0_row3_mul; stage0_row3_mod; }
    }
    // Update intermediary registers.
    par { precursor_0; precursor_1; precursor_2; precursor_3; }
    
    par {
      seq { stage1_row0_mul; stage1_row0_mod; }
      seq { stage1_row1_mul; stage1_row1_mod; }
      seq { stage1_row2_mul; stage1_row2_mod; }
      seq { stage1_row3_mul; stage1_row3_mod; }
    }
    // Write from registers back to memory.
    seq { epilogue_0; epilogue_1; epilogue_2; epilogue_3; }
  }
```
Let’s go through each of these steps.
### Preamble
The preamble essentially takes each value X in `a` and `phi` and writes it to a register `rX` and `phiX` respectively. This allows us to compute each stage in parallel. Before, we were blocked on the number of accesses to memory. This is done sequentially, so should take N cycles.

### Pipeline stage
This is the meat of the NTT algorithm. Here, we compute the necessary calculations for the butterfly unit. Since each calculation requires a multiply followed by a combinational `add` or `sub` with modulus, we split each stage into two pieces:
```
seq { 
  stage0_row0_mul;    // V = a[j] * phi[k]
  stage0_row0_mod;   // (U +- V) % q
}
```
[... 345 lines cut ...]
      }
      seq {
        stage1_row1_mul;
        stage1_row1_mod;
      }
      seq {
        stage1_row2_mul;
        stage1_row2_mod;
      }
      seq {
        stage1_row3_mul;
        stage1_row3_mod;
      }
    }
    // Write from registers back to memory.
    seq {
      epilogue_0;
      epilogue_1;
      epilogue_2;
      epilogue_3;
    }
  }
} // End control
} // End main
```

cgyurgyik, 2021-01-23:
cc @rachitnigam since you've probably already answered and/or solved this.

So I’m doing something like this currently for a multiply followed by modulus:
```
group do_mul_op {
  ...
  mul_temp.in = mult_pipe.out; // Save product in a register.
}

group do_mod_op {
  ...
  mod_pipe.right = mul_temp.out; // Use product.
}

// Control
...
seq { do_mul_op; do_mod_op; }
```
However, this does require using a temporary register between the multiply and the modulus operations. 

Obviously it’d be great to avoid using a temporary register here. Just using `mult_pipe.out` in `do_mod_op` won’t work since the signal isn’t visible outside the group. I think this is exactly why the `invoke` operator was created. Don’t I want to do something like:
```
group do_mod_op {
  mod_pipe.right = mult0.out; // Use product; avoids a temporary register.
}

...

// Control
seq { 
  invoke mult0(...)();
  do_mod_op;
}
```

In fact, we could probably just do the same thing with the group `do_mod_op` as well, and feed the output of `mult0` to the arguments of `mod0`. However, I still need to save the output of `do_mod_op` to a register for use in the next stage, so a group after “both” invokes is still required.

rachitnigam, 2021-01-24:
TLDR: Your approach is almost the best we can do in this case. The only possible optimization here to remove the register b/w mult and mod is sequence them within the group. 

You've run into two problems here:
1. The interface for `mult` and `mod` is not `invoke`-able (#331). I'm going to refer you to the issue for an explanation about this.
2. The compiler cannot reason about how long the signal from `mult` is alive for.

The hand optimization that you can do here—to eliminate the register b/w mult and mod would look like this:
```
group mult_mod_fused {
  // Run mult exactly once. mult.go = !mult.done is going to repeatedly run mult.
  mult.go = !once.out;
  once.out = mult.done;
  once.write_en = mult.done;
  // args to mult ...

  // Start mod on the cycle when the signal from mult is visible
  mod.go = mult.done || once.out;
  mod.left = mult.out; // this signal is valid on the first cycle. After that mod "saves" it.
  mod.right = ...
  mult_mod_fuse[done] = mod.done;
}
```

The compiler cannot reason about this optimization. It does not have the following pieces of information:
1. `mult.done` is alive for the same cycle as `done`. The compiler always has to assume that signals output signals are dead.
2. `mod.left` is *only* needs to be driven for one cycle. The `mod` component "saves" the signal after that cycle for use.

All of this goes back to the discussion we were having on Tuesday about the liveness of signals. The is *mostly* ignorant of which signals are alive when. The one thing it does know is that the `out` signal of registers is always alive. An extension to the Calyx compiler that reasons about the liveness of signals would allow these kinds of "chaining" optimizations. That's what we want to build next.

CC @sampsyo @sgpthomas @tedbauer since you might find this discussion interesting

cgyurgyik, 2021-01-24:
Thanks for the long write-up.

> `mult.done` is alive for the same cycle as `done`. The compiler always has to assume that signals output signals are dead.

What is the second `done` you're referring to?

Ok, so I was assuming that `mult` and `mod` are` invoke`-able. #331 says that is not the case. it seems that problem 2, signal aliveness, is solved if we make it so?

One thing that I didn't mention here is that each stage only requires `N/2` multiplications, since there are duplicates.
Eventually, I only want to use `N / 2` `mult_pipe`s, which means that I'll want the signal connected to two different `mod_pipe`s. By guaranteeing the signal will last after invocation, and placing it before said `mod_pipe`s in a `seq`, this seems correct.

rachitnigam, 2021-01-24:
Ah, typo.  I meant to say: `mult.out` is available when `mult.done` is high.

> Ok, so I was assuming that mult and mod are invoke-able. #331 says that is not the case. it seems that problem 2, signal aliveness, is solved if we make it so?

The "solution" to #331 is introducing a register into the `mult_pipe` primitive. Morally, this doesn't fix your problem of extra registers. It just gives you a new interface that lets you use values after `invoke`-ing a primitive. The solution that lets us implement the hand written optimization I showed you in the compiler requires more work.

cgyurgyik, 2021-01-24:
A few more questions to digest this in its entirety.
1. In your example above, isn't `once` an extra  1-bit register as well?
2. You said:
> mod.left is only needs to be driven for one cycle. 

Curiously, is there anything wrong with "driving" it for longer? I'd assume no, since we write these to a register within the primitive.

Thanks for the responses!!

rachitnigam, 2021-01-24:
> In your example above, isn't once an extra 1-bit register as well?

Yes

> Curiously, is there anything wrong with "driving" it for longer? I'd assume no, since we write these to a register within the primitive.

No. This is what happens currently. The question is what can you *assume* given a black-box component. Right now, all components using another component have to assume:
1. All input signals need to be driven as long as `go` is driven
2. No output signals are available after `done` signal is high

The `invoke` operator gives a very weak guarantee: it says that it's on the frontend to assume that the output ports it's reading from after the `invoke` are not dead.

closed 2021-01-26 by rachitnigam

sampsyo, 2021-02-07:
Just wanted to say that this conversation I unfortunately read much later was super interesting! Indeed, this is a great motivator for more nuanced reasoning about what signals are available when as part of the "calling convention" for statically timed components…