calyxir/calyx #366: [ntt] Reduce number of `mult_pipe`s

cgyurgyik, 2021-01-26:
Currently, we use N `mult_pipe`s in each stage of the algorithm.

However, since each product is used in two different rows, we can reduce this to N / 2. We have:

```
a[k] = (U + V) % q
a[k+t] = (U - V) % q
```
Here, V is the product `a[k] * phis[x]`.

The implementation presently has:
```
par {
  seq { s0_r0_mult; s0_r0_mod; } 
  …
} 
```

Instead of parallel control of `seq`s, we want sequential control of `par`s. Specifically, 
```
seq { 
  par {
    // Save product to registers mul0…mul(N/2 - 1).
    mul0; mul1; mul2; … mul(N/2 - 1);
  } 
  par {
    // Use `mul` registers in addition or subtraction mod N.
    mod0; mod1; … mod(N);
  }
} 
```

This reduces the number of `mul` registers from N to N/2, and the number of `mult_pipe`s from N to N/2.

rachitnigam, 2021-01-26:
I don't follow why the `par` of `seq` version uses more multipliers than the `seq` of `par` version.

cgyurgyik, 2021-01-26:
Because the `par` of `seq` version (what we currently have) is this:
```
par {
  seq { s0_r0_mult; s0_r0_mod; } 
  seq { s1_r1_mult; s1_r1_mod; } 
  …
  // N times
} 
```
This means we have N `mult_pipe`s. Each one calculates the value `V` above on its own, in `sX_rY_mult`. 

Instead, we can calculate the N/2 `V` values before (in parallel), and then do the second part, addition or subtraction mod N (in parallel).

Does this make sense?

rachitnigam, 2021-01-26:
> However, this value V is the same for a given pair of rows in a stage.

What does this look like in the control code? I don't know the verbiage about the "stages" and "rows" so I can't visualize this.

cgyurgyik, 2021-01-26:
So if you look at the example in #358, we have the following table of calculations for `N` = 4:
```
// +---+-----------------------+-----------------------+
// | a |        Stage 0        |        Stage 1        |
// +---+-----------------------+-----------------------+
// | 0 | a[0] + a[2] * phis[1] | a[0] + a[1] * phis[2] |
// | 1 | a[1] + a[3] * phis[1] | a[0] - a[1] * phis[2] |
// | 2 | a[0] - a[2] * phis[1] | a[2] + a[3] * phis[3] |
// | 3 | a[1] - a[3] * phis[1] | a[2] - a[3] * phis[3] |
// +---+-----------------------+-----------------------+
```
There are log2(N) stages, and for each stage, in the pipeline, there are N rows. These stages must be calculated sequentially, and the rows can be calculated in parallel. Each row `i` represents the value of `a[i]` at the end of the stage. Not pictured here is `mod q`, but that's also part of each calculation. You'll notice that during stage 0, `a[0]` and `a[2]` both calculate `V = a[2] * phis[1]`. Similar for `a[1]`, `a[3]`.

The control for this currently is:
```
     ...
     // Stage 0
     par {
        seq { s0_r0_mul; s0_r0_mod; } // Calculate V, then U + V mod q
        seq { s0_r1_mul; s0_r1_mod; } 
        seq { s0_r2_mul; s0_r2_mod; } // Calculate V again, then U - V mod q 
        seq { s0_r3_mul; s0_r3_mod; }
      }
      // Load values into intermediary registers for parallel use.
      par { precursor_0; precursor_1; precursor_2; precursor_3; }

      // Stage 1
      par {
        seq { s1_r0_mul; s1_r0_mod; }
        seq { s1_r1_mul; s1_r1_mod; }
        seq { s1_r2_mul; s1_r2_mod; }
        seq { s1_r3_mul; s1_r3_mod; }
      }
      ...
```

Instead, we can change this to calculate all the `V`s in a stage first (in parallel), and then calculate all the addition / subtractions mod `q`. 

So what we have now is:
```
// Stage 0
seq {
  par { mult_V0; mult_V1; } // Calculate the N/2 unique `V` values and saves them to temporary registers mul0, mul1.
  par { 
    mod0; // (a[0] + mul0)   mod q
    mod1; // (a[1] + mul1)   mod q
    mod2; // (a[0] - mul0)   mod q
    mod3; // (a[1] - mul1)    mod q
 }
}

// Stage 1 is similar.
```

The difference here is we need to correctly hook up the calculated `V`s outputted in the first parallel control to the `mod`s calculates in the second parallel control

cgyurgyik, 2021-01-26:
TLDR: Each stage has at most `N / 2` unique products that I've named `V`. Before, we were calculating each unique `V` twice. We can avoid this by calculating the unique `V` values first.

closed 2021-01-27 by cgyurgyik