calyxir/calyx #367: Running FuTIL on real life FGPAs

sgpthomas, 2021-01-26:
This is an issue to track the steps needed to get FuTIL running on an FGPA. For now, we are targetting the Alveo U50 card that is in Havarti. Hopefully, the process won't be drastically different for different FPGAs.

The plan of attack is to modify this [RTL VADD](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/rtl_kernels/rtl_vadd) example to work with a simple vector add written in FuTIL. I've got this example running on hardware in Havarti.

Here are a rough outline of the steps needed to get this done:
 - [ ] Figure out the AXI4 interface that the VADD example uses to read/write memory
 - [ ] Rewrite this (in Verilog) to use boring old memories (BRAMs?) instead of FIFOs
 - [ ] Try swapping out the vector add for something like a dot product
 - [ ] separate out the vector add / dot product implementation into Verilog that assumes a FuTIL like memory interface
   - [ ] Figure out if we need to change FuTIL memory interface 
 - [ ] use FuTIL to generate the kernel and use the AXI4 stuff as a wrapper
 - [ ] have FuTIL / something generate the AXI interfaces
 - [ ] Think about how to support streaming AXI?

sgpthomas, 2021-01-26:
Simple description of AXI4-lite with some useful diagrams and charts: [https://www.realdigital.org/doc/a9fee931f7a172423e1ba73f66ca4081](https://www.realdigital.org/doc/a9fee931f7a172423e1ba73f66ca4081).

Some acronyms:
 - `*_hs`: hand shake
 - `gie`: global interrupt enable
 - `ier`: interrupt enable register
 - `isr`: interrupt status register

From this document: https://www.xilinx.com/support/documentation/ip_documentation/axi_intc/v4_1/pg099-axi-intc.pdf

sgpthomas, 2021-01-26:
some more introduction to axi articles

https://forums.xilinx.com/t5/Design-and-Debug-Techniques-Blog/AXI-Basics-1-Introduction-to-AXI/ba-p/1053914
https://forums.xilinx.com/t5/Design-and-Debug-Techniques-Blog/AXI-Basics-2-Simulating-AXI-interfaces-with-the-AXI-Verification/ba-p/1053935

also discovered that `.sv` files don't work with the IP integrator. There needs to be a Verilog (`.v`) wrapper file.

sgpthomas, 2021-02-03:
An easy short term solution is using `$readmemh` to embed data directly into a bitstream instead of using AXI to communicate. If we did this, I don't know how we get the memories back off the FPGA after the kernel runs.

sampsyo, 2021-02-03:
Haha, that's a good point—that does seem a lot harder than the `$readmemh` literal trick! Maybe we should make it blink an LED in Morse code.

sgpthomas, 2021-04-20:
Updating this with some new information.

# Kernel <-> Host Interfacing
In order to get openCL/XRT C host code to be able to run and interact with kernels, the kernel needs to implement the AXI4-lite slave interface. [Here](https://developer.arm.com/documentation/ihi0022/e/AMBA-AXI4-Lite-Interface-Specification/AMBA-AXI4-Lite/Definition-of-AXI4-Lite/Signal-list?lang=en) is where this interface is defined. [ZipCPU](https://zipcpu.com/formal/2018/12/28/axilite.html) has some good resources for understanding and implementing AXI4-lite (it also shows that the Xilinx reference implementation of AXI is broken and slow which fills be with dread and fear. but also the techniques used to show this are cool). 

## AXI
Let's start with a brief introduction to how AXI works. The basic idea, is that AXI4 has 5 separate channels for coordinating "reads" and "writes" (I put these in quotes because the reads/writes aren't necessarily from memory. They are abstracted terms for sending information and receiving information). It is addressed based, which means that you ask to read/write to some abstract address (again not necessarily a memory address).

The channels are:
 - Write Address Channel (prefix `aw`): Communicates addresses that should be written to.
 - Write Data channel (prefix `w`): Communicates data that should be written to the address communicated over the `aw` channel
 - Write response channel (prefix `b`): Communicates the a write has been successful
 - Read address channel (prefix `ar`): Communicates addresses that should be read from.
 - Read data channel (prefix `r`): Communicates the data that is read from the address received over the `ar` channel

Each of these channels have their own async `valid/ready` handshakes. So to send something along a channel, you set the data for the channel as well as the ready or valid (depending on whether you are master or slave) signal and wait till the other side sets their `ready/valid` signal.

The funky part about these channels, is that they aren't synchronized at all by the bus. What I mean is that the address and data are communicated over separate channels that are separately handshaked and must be synchronized internally by the slave or master (wish they used different terminology). In order to perform a write, three things must happen:
 1) send address over `aw` channel w/ handshake
 2) send data over `w` channel w/ handshake
 3) watch `b` channel to see when the write is received w/ handshake

The reason for all this complexity is to support _bursts_, i.e. writing / reading a bunch of data at once. For the time being, I will complete ignore this extra complexity and will gladly accept slowness as my punishment.

## Kernel requirements
This information comes from here: https://xilinx.github.io/Vitis-Tutorials/master/docs/Hardware_Accelerators/Feature_Tutorials/01-rtl_kernel_workflow/README.html#hardware-interface-requirements

In order for host code on the CPU to be able to invoke our kernel, we need to implement an AXI4-lite slave interface. This interface has the following address space:
 - `0x00`
   - bit 0: `ap_start` (Host writes here to start the kernel)
   - bit 1: `ap_done` (kernel write here to signal to host it's done)
   - bit 2: `ap_idle` (kernel writes here to signal it is ready to be run again)
 - `0x04 - 0x0c`
   - various interrupt registers that I don't entirely understand
 - `0x10`: The value of a scalar argument. For some reason this is required even if you don't need to take in a scalar argument.
 - `0x14`: reserved for some non-sense presumably
 - `0x18-\infty`: pointer to array values in global memory. This is the main way to transfer data to and from an accelerator.

The CPU first writes the scalar and pointer arguments, then sets `ap_start`. This means the kernel can assume that the arguments are present when it starts running.

Besides the slave interface that let's Xilinx tell the kernel to start, you also can implement several AXI4 masters to communicate with memory. The model is that the memory is an AXI4-slave and so you can implement a master to write/read data from it. From some experimentation, it seems that you can write a very simple AXI4 master (no bursting, one operation at a time) to operate memory. This wasn't immediately obvious because their examples implement some amount of bursting to make things go faster.

# Next steps
## AXI4 slave control interface
The first step is to generate a wrapper over the toplevel Calyx component that implements the control interface and passes
arguments to the toplevel component. As a first pass, we should implement the wrapper in Verilog (not SystemVerilog because Vitis doesn't accept this for toplevel modules). Because arguments change, this wrapper will need to use that `external(1)` annotation to parametrically generate this wrapper. Fortunately, this seems pretty easy to do (it's easy to add more arguments to the interface without touching the rest of the wrapper).

For a second pass it would be cool to represent the AXI interface directly in Calyx. However, Calyx can not easily represent the multi-channel control interface of AXI, and so this will require some deep (but interesting, I hope) thinking. If we do this right, we could support different classes of AXI interfaces for different requirements easily as well as making the AXI code much much easier to read.

## Memory interface translation
[... cut ...]
All these files depend on Calyx external memory definitions. Right now, the compiler is the only place where this information lives. This means that the generation of these files needs to be in the compiler, or we need to duplicate the information somewhere else. One option is to have the compiler generate a JSON metadata file that has this information so that other tools (`fud`) can process it. We can also just have this in the compiler.

The first two files are very templately and should be easy to generate. The host code is potentially more complicated, but it would be cool if we could support the same JSON format that we use for simulation.

sampsyo, 2021-04-21:
Wow; this is super helpful, @sgpthomas! It's definitely the clearest that I have ever understood AXI before. :smiley: Here are a couple of small clarifying questions, in case you happen to know the answer…

To check my understanding, by this:

> The funky part about these channels, is that they aren't synchronized at all by the bus. What I mean is that the address and data are communicated over separate channels that are separately handshaked and must be synchronized internally

Do you mean that it does *not* work so that you could just set `aw` and `w` simultaneously with a single handshake, or something like that? (Which would have seemed sensible to me.) Instead, you have to "negotiate" the "write data" and "write enable" signals entirely separately, which is pretty weird but maybe not too bad.

And here's a silly question:

> `0x00`
> * bit 0: `ap_start` (Host writes here to start the kernel)
> * bit 1: `ap_done` (kernel write here to signal to host it's done)
> * bit 2: `ap_idle` (kernel writes here to signal it is ready to be run again)

Maybe I was misunderstanding, but I thought that, in this setup, the kernel couldn't write anything—only the host could do that? So does the host just busy-wait by repeatedly reading bit 1 of byte 0, or is there some way I missed for the kernel to "push" this state change to the host when it's finished?

---

And now about the concrete next steps. I like the idea of implementing some crud in plain old Verilog as much as possible at first, and worrying later about how to expose more and more of this interface stuff "in Calyx" farther down the line.

That is, the most monolithic (i.e., simplest from Calyx's perspective) approach I can imagine is that a big blob of Verilog is responsible for communicating with the host, filling up some BRAMs, sending a start signal to the actual Calyx-generated hardware, receiving its done signal, relaying that back to the host, and servicing the host's read requests for data in the BRAMs. So the interface between the Verilog blob and Calyx would consist of externalized memory interfaces and a top-level `go`/`done` pair—nothing more. Is that what you're imagining for the first step?

It's also a good question about where all the generation stuff (which is Xilinx-specific, obviously) should "live." It would be nice to decouple it from the compiler… like you say, it would be cool if the same format also worked for simulation. Conceptually speaking, it feels like the JSON format you're imagining is the moral equivalent of an interface declaration (like a C header or an `.mli` file in OCaml). It would be conceptually nice if there could be a standalone, Calyx-independent module of some kind (regardless of whether it's written in Rust and linked with the compiler or an actually separate program, like fud) that _just_ needed this interface spec to generate all the Xilinx-specific crud. Maybe we should think a little more about how to "standardize" this interface declaration format so it's useful for Xilinx interfacing, simulation, and maybe other interfaces in the future?

sgpthomas, 2021-04-21:
> Do you mean that it does not work so that you could just set aw and w simultaneously with a single handshake, or something like that? (Which would have seemed sensible to me.) Instead, you have to [...]
Yes this is exactly what I mean. Otherwise there would not be a reason for these signals to be separately handshaked (is that a word?). The reason for this, as far as I can tell, is to support bursting, i.e doing a bunch of writing and reading in a row without waiting for the response. Though, I'm not certain how this is supposed to work.  However, for the -lite version of the protocol, Xilinx does seem to set some practical bounds: the write address and write data can't be separated by more than a cycle. This doesn't seem to be a part of the spec though.

> Maybe I was misunderstanding, but I thought that, in this setup, the kernel couldn't write anything—only the host could do that? So does the host just busy-wait by repeatedly reading bit 1 of byte [...]
The short answer is yes, the host just busy-waits reading bit 2 (not bit 1) to see if the kernel is done. The way to think about this is that the kernel (an axi slave) is pretending to be a memory. The host (axi master) can read / write to the "memory" as normal, but because the kernel isn't actually a memory, it can sneakily change the value at some addresses. When the host and kernel agree on the meaning of addresses, this mechanism facilitates communication. But the communication must be driven by reads / writes from the master. It feels analogous to parallel processes communicating by writing / reading from shared memory.

I like the idea of standardizing the interface declaration stuff. I'll think a bit about what this should look like.

sampsyo, 2021-04-21:
Got it; thanks!! Just to sum up with what's going on with "shared memory," maybe the right buzzword analogy is ["memory-mapped I/O,"](https://en.wikipedia.org/wiki/Memory-mapped_I/O) i.e., special addresses in memory that do things other than literally reading/writing some memory.

> handshaked (is that a word?)

Handshaken? 😂

rachitnigam, 2021-05-25:
@sgpthomas does #492 close this issue?

closed 2021-05-25 by sgpthomas