calyxir/calyx #372: Unify components and primitives

rachitnigam, 2021-01-28:
Let's do this in three steps:

### Change component instantiation syntax
In the quest to eventually unify primitives and components, we should make the syntax of components look like:
```
cells {
  c = counter();
  r = prim std_reg(32);
}
```

### Remove the `prim` keyword
Get rid of the `prim` keyword as well. 

### Add parameters to component definition without making them usable
Add parameter syntax to component without allowing their use in the body of a Calyx component. This would let us remove `primitive` altogether and in the future add support for parametric components.

sampsyo, 2021-01-28:
Great. Just to be clear, this can all happen without "hooking up" those Calyx-component parameters to any sort of metaprogramming system. That is, we can totally kick the metaprogramming can down the road while getting this very nice unification.

rachitnigam, 2021-01-28:
Yup, that's exactly the idea. In fact, all of the steps can (and probably should) be performed as individual PRs.

closed 2021-01-31 by rachitnigam