calyxir/calyx #374: MrXL correctness tests don't pass

cgyurgyik, 2021-01-28:
With ` runt -i frontends/mrxl/test -d`
```
mrxl correctness (3 tests)
  ✓ pass - frontends/mrxl/test/add.mrxl
  ? miss - frontends/mrxl/test/dot.mrxl
         ~
        1│+ ---STDERR---
        2│+ ERROR: Stage 'futil' had a non-zero exit code.
        3│+ --------------- stderr ---------------
        4│+ Error:
        5│+ 76 |  dot_b0.write_en = 1'b1;
        6│+    |  ^^^^^^ Undefined cell name: dot_b0
        7│+ 
        8│+ 
         ~
  ? miss - frontends/mrxl/test/sum.mrxl
         ~
        1│+ ---STDERR---
        2│+ ERROR: Stage 'mrxl' had a non-zero exit code.
        3│+ --------------- stderr ---------------
        4│+ Traceback (most recent call last):
        5│+   File "/opt/anaconda3/bin/mrxl", line 8, in <module>
        6│+     sys.exit(main())
        7│+   File "/opt/anaconda3/lib/python3.8/site-packages/mrxl/main.py", line 32, in main
        8│+     emit(ast)
        9│+   File "/opt/anaconda3/lib/python3.8/site-packages/mrxl/gen_futil.py", line 294, in emit
       10│+     cells.append(emit_mem_decl(decl.name, decl.type.size, name2par[decl.name]))
       11│+ KeyError: 'foo'
       12│+ 
       13│+ 
         ~

```

Also, the [documentation](https://capra.cs.cornell.edu/calyx/frontends/mrxl.html) is outdated, as there is no `dot.json`, or any `.json` files (`.data` is used instead) in the `test directory. In fact, there is no `dot.mrxl.data` either.

rachitnigam, 2021-01-28:
Yeah, the initial commit to the compiler didn't fix these issues. Feel free to either ignore them or fix them.

rachitnigam, 2021-03-29:
@cgyurgyik  @sgpthomas @sampsyo: I'm not sure how like this issue being fixed is. None of us "own" the MrXL frontend and, since it's not a primary frontend, have any incentive to fix this. What should we do?

sampsyo, 2021-03-29:
I suppose I don't mind leaving it available in case someone has a moment to help with it, perhaps as a warmup? The goal is pretty clearly defined, so it seems OK to leave the ticket open to indicate the available work.

cgyurgyik, 2021-03-29:
Agreed with Adrian. I think this would be under the "Good First Issue" tag, for someone in the research group or someone just interested in Calyx. I think that tag would be neat for future issues as well, so that it is a bit more obvious which issues are beginner-friendly.

closed 2022-12-03 by rachitnigam