calyxir/calyx #389: Examples fail

calewis, 2021-02-03:
I have followed the build instructions and everything seems to build okay
```console
 cargo run -- --help                                                                                    
    Finished dev [unoptimized + debuginfo] target(s) in 0.13s
     Running `target/debug/futil --help`
futil 0.1.0
The FuTIL Authors

USAGE:
    futil [FLAGS] [OPTIONS] [--] [file]

FLAGS:
        --force-color
        --debug          Enable debug mode output
        --synthesis      Enable synthesis mode
    -h, --help           Prints help information
        --list-passes    list all avaliable pass options
    -V, --version        Prints version information

OPTIONS:
    -b, --backend <backend>                 Select a backend [default: futil]
    -d, --disable-pass <disable-pass>...    Disable pass during execution
    -l, --lib-path <lib-path>               Path to the primitives library [default: .]
    -o, --output <output>                   Output file [default: -]
    -p, --pass <pass>...                    Run this pass during execution [default: all]
    -t, --toplevel <toplevel>               Toplevel component [default: main]

ARGS:
    <file>    Input futil program
```

But when I try to run the examples I get an error
```console
 cargo run -- examples/futil/simple.futil -b verilog                                                     
    Finished dev [unoptimized + debuginfo] target(s) in 0.06s
     Running `target/debug/futil examples/futil/simple.futil -b verilog`
Error: FuTIL Parser:  --> examples/futil/simple.futil:9:19
  |
9 |     const0 = prim std_const(32, 4); // 32-bit width constant with value 4.␊
  |                   ^---
  |
  = expected args
```

If I run the tests using runt I get 19 failing tests, maybe this is connected? 
```console
 runt --exclude frontend   
passes (33 tests)
  ✓ pass - tests/passes/collapse-control.futil
  ✓ pass - tests/passes/compile-empty.futil
  ✓ pass - tests/passes/compile-invoke.futil
  ✓ pass - tests/passes/component-interface.futil
  ✓ pass - tests/passes/dead-cell-removal.futil
  ✓ pass - tests/passes/externalize.futil
  ✓ pass - tests/passes/go-insertion.futil
  ✓ pass - tests/passes/inliner.futil
  ✓ pass - tests/passes/merge-assign.futil
  ✓ pass - tests/passes/infer-static/component.futil
  ✓ pass - tests/passes/infer-static/groups.futil
  ✓ pass - tests/passes/infer-static/invoke.futil
  ✓ pass - tests/passes/resource-sharing/cond-port.futil
  ✓ pass - tests/passes/resource-sharing/multiple-cells.futil
  ✓ pass - tests/passes/resource-sharing/rename-swap.futil
  ✓ pass - tests/passes/resource-sharing/share.futil
  ✓ pass - tests/passes/minimize-regs/continuous-assignment.futil
  ✓ pass - tests/passes/minimize-regs/escape-boundary.futil
  ✓ pass - tests/passes/minimize-regs/live-register-analysis.futil
  ✓ pass - tests/passes/minimize-regs/nested-par.futil
  ✓ pass - tests/passes/minimize-regs/par-while-liveness.futil
  ✓ pass - tests/passes/minimize-regs/par-write.futil
  ✓ pass - tests/passes/minimize-regs/simple-liveness.futil
  ✓ pass - tests/passes/minimize-regs/thread-local.futil
  ✓ pass - tests/passes/compile-control/compile-if-static.futil
  ✓ pass - tests/passes/compile-control/compile-if.futil
  ✓ pass - tests/passes/compile-control/compile-par-static.futil
  ✓ pass - tests/passes/compile-control/compile-par.futil
  ✓ pass - tests/passes/compile-control/compile-seq-static.futil
  ✓ pass - tests/passes/compile-control/compile-seq.futil
  ✓ pass - tests/passes/compile-control/compile-while-static.futil
  ✓ pass - tests/passes/compile-control/compile-while.futil
  ✓ pass - tests/passes/regressions/group-multi-drive.futil
parsing (2 tests)
  ✓ pass - tests/parsing/attributes.expect
  ✓ pass - tests/parsing/guards.expect
errors (12 tests)
  ✓ pass - tests/errors/duplicate-cells.futil
  ✓ pass - tests/errors/mismatch-widths.futil
  ✓ pass - tests/errors/no-drive.futil
  ✓ pass - tests/errors/papercut-cell-and-group-conflict.futil
  ✓ pass - tests/errors/papercut-cell-as-group.futil
  ✓ pass - tests/errors/papercut-no-done.futil
  ✓ pass - tests/errors/reserved-cell-name.futil
  ✓ pass - tests/errors/unused-group.futil
  ✓ pass - tests/errors/parser/cell-missing-semi.futil
  ✓ pass - tests/errors/parser/num-without-bitwidth.futil
  ✓ pass - tests/errors/parser/while-without-body.futil
  ✓ pass - tests/errors/parser/wrong-binary-num.futil
correctness dynamic (7 tests)
  ✗ fail - tests/correctness/diff_width_test.futil
  ✗ fail - tests/correctness/if.futil
  ✗ fail - tests/correctness/invoke.futil
  ✗ fail - tests/correctness/par.futil
  ✗ fail - tests/correctness/seq.futil
  ✗ fail - tests/correctness/udot_product.futil
  ✗ fail - tests/correctness/while.futil
correctness static timing (7 tests)
  ✗ fail - tests/correctness/diff_width_test.futil
  ✗ fail - tests/correctness/if.futil
  ✗ fail - tests/correctness/invoke.futil
  ✗ fail - tests/correctness/par.futil
  ✗ fail - tests/correctness/seq.futil
  ✗ fail - tests/correctness/udot_product.futil
  ✗ fail - tests/correctness/while.futil
systolic array correctness (3 tests)
  ✗ fail - tests/correctness/systolic/array-1.systolic
  ✗ fail - tests/correctness/systolic/array-2.systolic
  ✗ fail - tests/correctness/systolic/array-3.systolic
NTT pipeline correctness (2 tests)
  ✗ fail - tests/correctness/ntt-pipeline/ntt-16.txt
  ✗ fail - tests/correctness/ntt-pipeline/ntt-8.txt
verilog backend (1 tests)
  ? miss - tests/backend/verilog/dot-product.futil
Python library AST example (3 tests)
  ✓ pass - calyx-py/test/example.py
  ✓ pass - calyx-py/test/if.py
  ✓ pass - calyx-py/test/invoke.py

1 missing
19 failing
50 passing
```
Any advice on how to get the examples working?

sgpthomas, 2021-02-03:
There was a recent syntax change that removed the prim word. We forgot to update the examples. We'll remedy that A.S.A.P (and should probably test the examples)

sgpthomas, 2021-02-03:
The failling tests are probably due to something else though. Can you run `runt` with `-d` to see the failling output of those tests?

calewis, 2021-02-03:
```console
correctness dynamic (7 tests)
  ✗ fail - tests/correctness/diff_width_test.futil
         ~
    1    │- {
    2    │-   "mem0": [
    3    │-     [
    4    │-       12,
   ...
   80    │- }
   81    │-
        1│+ ---STDERR---
        2│+ sh: fud: command not found
        3│+
         ~
 ```
 I guess I didn't get fud installed correctly.

calewis, 2021-02-03:
```console
        1│+ ---STDERR---
        2│+ Traceback (most recent call last):
        3│+   File "/Users/canlewi/Software/Calyx/futil/frontends/ntt-pipeline/gen-ntt-pipeline.py", line 1, in <module>
        4│+     from prettytable import PrettyTable
        5│+ ModuleNotFoundError: No module named 'prettytable'
        6│+
```
I got `fud` into my path, but there seem to be some dependencies that don't automatically get installed.  I know it's crazy, but I didn't have numpy or some other common packages like prettytable installed on this computer. 

With prettytable installed I get:
```console
        1│+ ---STDERR---
        2│+ WARNING: global.futil_directory is not set in the configuration
        3│+ ERROR: 'global.futil_directory' is not set. Use `fud config {path_str} <val>` to set it.
        4│+
```

sgpthomas, 2021-02-03:
yeah we have many separately managed python projects and so the dependencies are a mess. we're working on that. 

about `fud`. this is documented here: https://capra.cs.cornell.edu/calyx/tools/fud.html. you can also type `fud config` and it will ask you for the global directory (we need to document this)

calewis, 2021-02-03:
Alright, I finally read all the doc `fud config global.futil_directory <full path to FuTIL repository>` and now running the tests works.

calewis, 2021-02-03:
> about `fud`. this is documented here: https://capra.cs.cornell.edu/calyx/tools/fud.html. you can also type `fud config` and it will ask you for the global directory (we need to document this)

It was there, I just jumped in without reading everything.  One minor suggestion for the doc: 
Include `fud config global.futil_directory <full path to FuTIL repository>` on the getting started page, that way if people don't go fully read the `fud` documentation, but do follow all the steps on the getting started page things should work.

cgyurgyik, 2021-02-03:
Hi @calewis , the examples should be up-to-date with #391. Thanks for pointing this out, and please feel free to ask any questions.

closed 2021-02-03 by cgyurgyik