calyxir/calyx #390: Improve initial setup for FuTIL

sgpthomas, 2021-02-03:
There are a lot of tools that need to be installed to get tests running properly. It would be nice to have a script / instructions with a concise list of commands to run to get everything running so that the initial setup process is simpler.

cgyurgyik, 2021-02-03:
I think this is along the same lines as #376 ?

sampsyo, 2021-02-04:
I think a quick reorganization of the [Getting Started](https://capra.cs.cornell.edu/calyx/) steps might be great! There are many options there, and it's not 100% clear from the current structure what you're supposed to do as a "user" of Calyx. For example, you probably _do_ want fud, but you probably _don't_ need the Python library installed unless you are doing something specific.

So we could just list, at the top, a sort of minimal walkthrough to getting basic stuff working (clone repo, cargo build, install deps, install fud). And then the later sections could describe what specific reasons you'd want to install other things.

PR #395 (2021-02-22): Rewrite Getting Started instructions
Rewrite the getting started instructions so that there is a core installation and then other components are broken off into self contained sub-sections. Tried to keep the document as self contained as possible so that you don't have to jump around to other pages.

Additionally separated out some "core" tests that work after just the minimal installation.

fixes #390, fixes #392
closed 2021-02-22 by sgpthomas