calyxir/calyx #401: Relay's `fromtext` throws exception for `expand_dims`

cgyurgyik, 2021-02-12:
## How Relay IR is emitted and lowered
There are two ways to emit Relay IR, and lower it to Calyx. 

1. Using the TVM Python library:

```
def expand_dims():
    x = relay.var('x', shape=[512], dtype='int32')
    return relay.Function([x], relay.expand_dims(x, axis=1, num_newaxis=2))

# A few unimportant steps are skipped here, which can
# be seen in frontends/relay/example.py
relay_ir = expand_dims()

print(
  emit_futil(relay_ir)
)
```

2. Using relay.fromtext():
```
relay_ir = relay.fromtext(
  f”””
    v0.0.4
    fn (%x: Tensor[(512), int32]) -> Tensor[(512, 1, 1), int32] {
      let %x1: Tensor[(512, 1, 1), int32] = expand_dims(%x, axis=1, num_newaxis=2);
      %x1
    }
  “””
)

print(
  emit_futil(relay_ir)
)
```

## expand_dims
So, there is one issue. [expand_dims](https://tvm.apache.org/docs/api/python/relay/index.html#tvm.relay.expand_dims) works fine when using the Python library. However, when trying to use `relay.fromtext`, we get the following error:

```
  ...
  res = self.call(func, args, attrs, [])
  File "/Users/cgyurgyik/.local/lib/python3.8/site-packages/tvm/relay/_parser.py", line 633, in call
    return func(args, attrs, type_args)
  File "/Users/cgyurgyik/.local/lib/python3.8/site-packages/tvm/relay/_parser.py", line 95, in __call__
    raise Exception("Operator {} is not registered. It's attributes are {}"
TVMError: Check failed: ObjectTypeChecker<TObjectRef>: :Check(ptr): Expect Attrs but get Map
During handling of the above exception, another exception occurred:

Exception: Operator expand_dims is not registered. It's attributes are {'axis': Constant(-1)}
```

## What we use and where
For testing, we currently write the relay IR by hand, and use `relay.fromtext`. Switching this to using the Python library wouldn’t be too much of a pain, and would allow us to test `expand_dims`.

For the examples, we use the Python TVM library.

I should explicitly mention we don’t need `expand_dims` for VGG Net, and even if we did, we could always just use the Python library to produce the code. I’m posting this mostly for clarity, and in case someone runs into it again.

## Steps to take
I haven’t taken much time to look into why this is occurring, as it doesn't really block me from anything, other than writing a test for `expand_dims`. Since the function is pretty simple, I wasn't too worried. We could potentially change our tests to use the Python library to produce Relay IR, rather than hand-writing it, though I don't think that's entirely necessary for now.

Trying to find similar issues on the main TVM repository did not yield anything useful. I *think* that `relay.fromtext` no longer exists in the main repository either? I'd need to re-confirm this.

sampsyo, 2021-02-12:
Seems annoying, even if it's not a huge deal!

About getting help from the actual TVM people… I do seem to recall hearing that they may have replaced/revamped the Relay text parser recently? So `fromtext` may indeed not be in the same place anymore. It does look like it's at least referenced on HEAD, though:
https://github.com/apache/tvm/blob/0d303b46cac4420f2b83ed3a0f445060d2fd5982/tests/python/relay/test_ir_parser.py#L84

It could be worth asking on the TVM Discourse if you're not sure it's a bug?

cgyurgyik, 2021-02-12:
> Seems annoying, even if it's not a huge deal!
>
> About getting help from the actual TVM people… I do seem to recall hearing that they may have replaced/revamped the Relay text parser recently? So [...]

Yeah my main concern was that we are using an older version of TVM. I imagine the response I would get is "Did you try it on the newest version 0.X?" If you think its still worth doing it / there's a better way to do it, I'm happy to.

sampsyo, 2021-02-12:
Yeah, good question. Not sure, but I guess it would be worth a shot just asking, even if actually filing a bug would be bad form. :smiley:

cgyurgyik, 2021-02-12:
Post made in the TVM Apache forum [here](https://discuss.tvm.apache.org/t/expand-dims-throws-exception-in-older-tvm-hash/9120).

rachitnigam, 2021-04-01:
Doesn't look like there were any responses to the TVM thread. Is there anything actionable we can do on this thread? If not, let's close it/triage it somewhere?

closed 2021-04-01 by cgyurgyik