calyxir/calyx #417: Register Unsharing

rachitnigam, 2021-03-07:
We all know and love our register allocation/register sharing pass. The goal of a register *unsharing* pass is finding values stored in the same registers with different lifetimes and allocating new registers for them. For example, for the following high-level program:
```
let x = 1;
x := x + 1;
let z = x + 1;
```

We can rewrite it to:
```
let x = 1;
y := x + 1;
let z = y + 1;
```

### Rationale
A reasonable question to ask is why would be ever want to create *more* registers? After all, optimization literature for traditional compilers teaches us to always minimize registers (although that might also not be the best idea). In hardware designs registers are physical resources. When you reuse a register for a new value, the generated hardware needs to generate additional *multiplexers*. This means register (un)sharing is a register-multiplexer trade-off. 

The second interesting fact is that ASIC (silicon chips) treat registers as an expensive resource and multiplexer as a cheap resources. However, FPGAs have the inverse relationship: registers are cheap while multiplexers are expensive. This means that register (un)sharing becomes a target-specific optimization instead of something that is always good.

### Implementation Strategy

I don't yet have a clear idea on how to do this. The `minimize-regs` pass calculates the lifetime of registers, not values (AFAIK), so building on that pass would be difficult. The ideal implementation would do the following: use an analysis to calculate lifetimes of values and use a heuristic-based pass to perform *both* sharing and unsharing.

EclecticGriffin, 2021-04-02:
A few more examples:

```
x = 0
if cond:
   x = 2
else:
   x = 3

z = x + 187
```

can be rewritten as

```
x = 0
if cond:
   y = 2
else:
   y = 3

z = y + 187
```

A one branch write however, does not permit such re-writing.

```
x = 0
if cond:
   x = x + 3

z = x - 3
```

Renaming any of the definitions of x here requires renaming all of them.

The same behavior can be observed in a while loop example.
```
x = 0 
while cond:
    x = x + 2

z = x -7
```

It seems that this problem is based on a reaching variable analysis and considering what groupings of definitions for a given variable can reach the same point in the program. For instance:

```
x = 0                \\ x1
if cond:
   x = 2             \\ x2
else:
   x = 3             \\x3

z = x + 187
```

x1 does not reach the definition of z, but x2 and x3 do so for this simple program its possible to rename either group though as a x2 and x3 can reach the same place, they must have the same name.

For the while example:
```
x = 0                    \\ x1 
while cond:
    x = x + 2          \\ x2

z = x -7
```

Both x1 (skipping the loop) and x2 (executing the loop at least once) reach the definition of z so while they can be renamed together there is no renaming that can separate the x1 and x2 definitions. A similar analysis holds for the single-if example above.

My loose intuition is that we're taking the program graph and partitioning the basic blocks based on reaching definitions for a given variable. Another framing could be that we're taking the set of definitions for a given variable name and partitioning them into "equivalence classes"---this isn't quite the right term, but captures my intuition---according to which definitions can reach the same location.

sampsyo, 2021-04-02:
Not to overcomplicate things, but in this example:

```
x = 0
if cond:
   x = 2
else:
   x = 3
z = x + 187
```

It is true to replace the last three occurrences of `x` with `y`, but it doesn't actually buy you much because `x = 0` is dead. I know a more complicated example would have *reads* to `x` in both of the `if` branches, but without those, renaming here just substitutes one variable for another and there are still two variables total in the whole program.

…unless, of course, we imagine some sort of "conditional move" to enable *more* unsharing, producing this:

```
x = 0
if cond:
   y1 = 2
   flag = 0
else:
   y2 = 3
   flag = 1
y = flag ? y1 : y2;  // "conditional move"
z = y + 187
```

Not sure if that's actually a good idea for hardware purposes, but I thought it was worth throwing into the mix.

rachitnigam, 2021-04-02:
The second example is worse because you need to mux both the two writes to `flag` and insert a new mux when selecting `y`. In contrast, renaming `x` in both branches to `y` only introduces one mux for the two possible writes to `y`:
```
x = 0
if cond:
  y = 1
else:
  y = 2
z = y + 187
```

closed 2021-05-24 by EclecticGriffin