calyxir/calyx #418: Static compilation of If may be wrong

rachitnigam, 2021-03-07:
The `static` annotation for an `if` is calculated as `max(tru,fal)` where `tru` is the latency of the true branch and `fal` is the latency of the false branch. However, the go/done interface requires that the go signal be kept high for exactly the latency of a group. This should mean that if one of the branches take way more time than the other, the `go` signal will cause a re-execution of the `if` block. 

The test case we currently have has branches of latency 2 and 3 which probably does catch this because after the execution of group, the internal state is reset after one cycle. We should probably try out an example with a branches with latencies 10 and 2.

sampsyo, 2021-03-07:
Makes sense overall! Just to clarify, calculating the `static` annotation for the `if` this way is not itself a problem, right? Instead, it's the way the generated control hardware works internally. It could (but probably doesn't currently) manage the two constituent `go` signals internally to respect the desired invariant—while still having an overall latency itself of `max(tru,fal)`.

rachitnigam, 2021-03-07:
Yeah, the latency annotation itself is not the problem. It is the generated hardware.

sgpthomas, 2021-03-15:
I don't think this is actually an issue.
I tried this program, which should increment `i` several times if this bug exists. 
```
import "primitives/std.lib";

component main() -> () {
  cells {
    eq = std_eq(1);

    // true register chain
    t0 = std_reg(32);
    t1 = std_reg(32);
    t2 = std_reg(32);
    t3 = std_reg(32);
    t4 = std_reg(32);

    // false branch
    i = std_mem_d1(32, 1, 1);
    add = std_add(32);
  }
  wires {
    group cond {
      eq.left = 1'd0;
      eq.right = 1'd1;
      cond[done] = 1'd1;
    }

    group true {
      t0.write_en = 1'd1;
      t1.write_en = t0.done;
      t2.write_en = t1.done;
      t3.write_en = t2.done;
      t4.write_en = t3.done;

      t0.in = 32'd1;
      t1.in = t0.out;
      t2.in = t1.out;
      t3.in = t2.out;
      t4.in = t3.out;
      true[done] = t4.done;
    }

    group false {
      add.left = i.read_data;
      add.right = 32'd1;
      i.write_data = add.out;
      i.write_en = 1'd1;
      i.addr0 = 1'd0;
      false[done] = i.done;
    }
  }
  control {
    if eq.out with cond {
      true;
    } else {
      false;
    }
  }
}
```
However, it does the correct thing. Digging a bit deeper this is the generated group compiling the if:
```
    group static_if<"static"=6> {
      cond[go] = fsm.out == 3'd0 & static_if[go] ? 1'd1;
      cond_stored.in = fsm.out == 3'd0 & static_if[go] ? eq.out;
      cond_stored.write_en = fsm.out == 3'd0 & static_if[go] ? 1'd1;
      false[go] = fsm.out > 3'd0 & fsm.out < 3'd2 & !cond_stored.out & static_if[go] ? 1'd1;
      fsm.in = fsm.out != 3'd6 & static_if[go] ? incr.out;
      fsm.write_en = fsm.out != 3'd6 & static_if[go] ? 1'd1;
      incr.left = static_if[go] ? fsm.out;
      incr.right = static_if[go] ? 3'd1;
      static_if[done] = fsm.out == 3'd6 ? 1'd1;
      true[go] = fsm.out > 3'd0 & fsm.out < 3'd6 & cond_stored.out & static_if[go] ? 1'd1;
    }
```
The go signal for `false` is held for the correct amount of time.

This makes me realize the static timing for `if` really only makes sense if the difference between the branch latencies is smaller than the dynamic overhead. This is an easy thing to check for with a pass.

sgpthomas, 2021-03-15:
I'll add this test to our suite.

closed 2021-03-15 by sgpthomas