calyxir/calyx #428: Provide debugging tips in the docs.

cgyurgyik, 2021-03-15:
This was something discussed in the meeting today. Perhaps a list of debugging tips may prove useful, but I'm not sure what to include, and how to formulate it i.e. perhaps an example using a specific waveform viewer?

## Tips
1. Lower to `vcd` and use a waveform viewer to determine where things may be going wrong.
2. Use `-vv` when lowering with `fud` to determine which stage is leading to failure.
3. ???

sgpthomas, 2021-03-16:
we should write this up more thoroughly, but here are a few:
 - compile to fud stage `futil-noinline` so that groups are still intact but you can find the FSM that corresponds to a group
 - compile to fud stage `vcd`, open in wave viewer, use FSM stage to orient where in the program execution you are
 - `fud e <file> --to vcd_json | jq .TOP.main.signal | unique` to quickly find if a signal ever changes

sgpthomas, 2021-03-31:
Started a debugging section in the docs. Let's update this when we think of new ways of debugging.

rachitnigam, 2021-04-01:
Is there a reason to keep this open? If https://github.com/cucapra/calyx/commit/89d801b8b68309e954d3bdce3eac112e9444da09 addresses the particular tips mentioned here, let's close this issue.

closed 2021-04-01 by sgpthomas