calyxir/calyx #429: Smarter static if compilation

sgpthomas, 2021-03-15:
Static compilation for `if` doesn't always reduce the latency of a component. For example consider:
```
@static(20) if c.out with cond {
  @static(1) true;
} else {
  @static(19) false;
}
```
The static timed version of this always takes 20 cycles whereas a dynamic version would take 2 cycles for the true branch and 21 cycles for the false branch.

Because of this, I think static timing only makes sense 
 - when the latencies of the branches are the same or off by 1
 - the if is in a sequence where cycles saved from static seq compilation > overhead for static if compilation

I propose we implement a pass that either
 - adds an annotation that tells the static compilation pass to ignore a control statement, or 
 - removes a static annotation from a control statement
using the above heuristic.

rachitnigam, 2021-03-15:
Also consider that a top down static compilation pass (#159) will not have this problem—it can always transition to the right state after the branch is done instead of taking this unnecessary extra time.

sgpthomas, 2021-03-16:
I'm not seeing why the top down version would be any different. The problem is that you can't statically know which branch will be taken so you have to assume that the longer branch is taken. You need to use the runtime value of the condition to dynamically choose which state to transition to. AFAIK this is also a problem for statically scheduled HLS right.

rachitnigam, 2021-03-16:
The problem with HLS branches usually has to do with branches in pipelined code. Top down CC can build an FSM that conditionally transitions from the branches as soon as they are done. Since the branch parent knows it’s running an if, it can use the timing of branch to transition early.

sgpthomas, 2021-03-23:
Subsumed by #159

closed 2021-03-23 by sgpthomas