calyxir/calyx #434: Numeric Type Testing

cgyurgyik, 2021-03-18:
## Fixed point
- [ ] Overflow `ufix<2,1>: 2b'1.0 + 2b'1.0 = 2b'0.0`
- [ ] Underflow `ufix<2, 1>: 2b'0.1 * 2b'0.1 = 2b'0.01`
- [ ] Numbers not representable by fixed point `0.1`
- [ ] (Signed only) Operations with result being negative `1.0 + (-1.5) = -0.5`
- [ ] (Signed only) Operations with result being positive `-1.0 + 1.5 = 0.5`
- [ ] Different-width overflow `ufix<2,1>: 2b'1.0 + 2b'1.0 = ufix<3, 1> 2b'10.0`
- [ ] Constants
- [ ] Add, substract
- [ ] Synthesizable `mult` and `div` pipes.
- [ ] Softmax operator (would be awesome).

## Bitnum
- [ ] Overflow `ubit<1>: 1'b1 + 1'b1 = 0'b1` 
- [ ] (Signed only) Operations with result being negative `1 + -2 = -1`
- [ ] (Signed only) Operations with result being positive `-1 + 2 = 1`

rachitnigam, 2021-03-18:
Python's random testing system Hypothesis might be useful for this: https://hypothesis.readthedocs.io/en/latest/

sampsyo, 2021-03-18:
Testing this stuff seems like a great idea!

Thinking about libraries like Hypothesis, it does occur to me that doing so would require a Python-y way to invoke compilation and simulation. One possible way to do this that _might or might not_ be a good idea is to give fud a proper Python API so it can be invoked directly from a test function—instead of just via the command line.

sgpthomas, 2021-03-18:
I think extending `fud` with a proper Python API could be cool and could enable _meta_ fud tools to be developed

rachitnigam, 2021-03-18:
To be clear, we can also just test json_dat conversation stuff with random testing and use normal testing on the rest of the stuff.

sampsyo, 2021-03-18:
Yeah, good call!

More unhelpful broad pondering: it seems like more people should want to use our JSON data format beyond just us. A well-specified, human-writable JSON format supporting a variety of numerical encodings that translates to `$readmemh`-compatible data files is a super good idea, and I am surprised that it doesn't already exist! (I have googled a bit and found nothing.)

closed 2021-04-01 by cgyurgyik