calyxir/calyx #435: Fixed point signed adder is incorrect.

cgyurgyik, 2021-03-18:
Given fixed point form: `Q6.3`, i.e.
Total width: 6, Integer width: 3, Fractional width: 3

We have:
```
   -1.5  101100
+   1.0  001000
_______________
   -0.5  100100
```

However, using the current [fixed point signed add](https://github.com/cucapra/calyx/blob/88612fb9407d731743e236550e65d38497a80054/primitives/fixed/signed.sv#L1-L11), we get:
`-2.5: 110100`, since its quite literally adding bits.

cgyurgyik, 2021-03-18:
Hmm perhaps this is because I'm misinterpreting it...

cgyurgyik, 2021-03-18:
I think this is an issue of interpretation. From my understanding, and how we are currently parsing fixed points, is that bit `0` is the signed bit, and bits `1:N` are the actual number. That means `add` will really need to conduct different calculations depending on whether the operands are (1) both positive, (2) both negative, or (3) different signs.

cgyurgyik, 2021-03-18:
If we instead interpret it as 2s complement fixed point,

Given that `0.5 = 000.100`

To get `-0.5`, we:
1. Invert bits: `111.011`
2. Add 0'b1: `111.100`

We need to adjust `-1.5` as well.
`1.5 = 001.100`
1. Invert bits: `110.011`
2. Add one: `110.100`.

Then we have:
```
// 2s complement
   -1.5  110.100
+   1.0  001.000
_______________
   -0.5  111.100
```

sampsyo, 2021-03-19:
Huh! Yeah, I totally agree that two's complement seems like the natural thing to do. The magical thing is that you just get to use a plain old integer adder and it works out, right? This whole two's complement idea is a seriously good innovation!!! 🥈😄

Pretty weird that the other library you found uses sign/magnitude. The whole "four cases for each of the four possible sign combinations" thing is a telltale sign that it seems more complicated than it needs to be. I wonder why… is there any reason why two's complement is better for integers than for fixed-point numbers? Can't think of one at the moment.

cgyurgyik, 2021-03-19:
- Decision: Use two's complement to represent FP.
- Reasoning: No concrete reason for why signed magnitude may be useful to represent fixed point numbers. Twos complement provides awesome [representation](https://en.wikipedia.org/wiki/Two%27s_complement) features, and parallels nicely with the bitnum representation.
- Changes required: Calyx just interprets bits -- not much there other than understand that the `fp_const` must use twos complement. However, the `dat` stage of fud currently interprets FP as signed magnitude. This needs to be changed to twos complement.

PR #445 (2021-03-23): Use two's complement in fixed point and improve precision.
Closes #435. Fixed point numbers are now parsed as two's complement.

In addition, this PR removes the use of `float` when calculating the fixed point number. Instead, a combination of Python's `Decimal` and `Fraction` are used to ensure that precise numbers are calculated. To verify this, `Hypothesis` tests are provided that round trips floating point numbers with bit widths in the range [2, 256].
closed 2021-03-23 by cgyurgyik