calyxir/calyx #442: minimize-regs pass is probably broken for uses of registers in conditions of `if` and `while`

sgpthomas, 2021-03-22:
The minimize regs pass doesn't rewrite the conditions of `if` and `while` loops. This means that if a register is used in the condition that was rewritten to something else, the program will no longer be correct because the register won't be updated in the way that is expected.

We should
 1) add a test case that checks for this (and verify that this is actually a problem)
 2) fix minimize regs to rewrite conds
 3) maybe implement a more general cell rewriting helper into the builder that does this automatically because this will be a problem for all passes that want to rewrite components

sgpthomas, 2021-03-22:
verified the bug. here's a program that breaks (and causes the compiler to crash)
```
// -p minimize-regs -p dead-cell-removal
import "primitives/std.lib";
component main() -> () {
  cells {
    x = std_reg(1);
    y = std_reg(1);
  }

  wires {
    group wr_x {
      x.in = 1'd1;
      x.write_en = 1'd1;
      wr_x[done] = x.done;
    }

    group cond {
      y.in = 1'd1;
      y.write_en = 1'd1;
      cond[done] = y.done;
    }

    group some_math {
      y.in = 1'd0;
      y.write_en = 1'd1;
      some_math[done] = y.done;
    }
  }

  control {
    seq {
      wr_x;
      if y.out with cond {
        some_math;
      }
    }
  }
}
```

sampsyo, 2021-03-23:
Good find! And nice work concocting a test case.

commit 5f3c1f2b (2021-03-23): Merge resource sharing and minimize regs (#444)

* add test for minimize regs rewriting conditions

* changing trait bounds + improved doc comments

* add `sharing_component.rs` to abstract common details between minimize-regs and
resource-sharing. fixes #442

* documentation + clippy fixes

* [fud] Errors in Fud prints out STDOUT

* Attempt to serialize workflows (#432)

* Give @external semantics for verilog backend (#430)

* Give @external semantics for verilog backend

@external(1) on a memory makes the backend generate code to initialize
memories and dump out their final states. Every oth [... cut]
closed 2021-03-23 by sgpthomas