calyxir/calyx #448: Dead cell removal removes cells that are used in invoke

sgpthomas, 2021-03-24:
This program:
```
import "primitives/std.lib";

component add(left: 32, right: 32) -> (out: 32) {
  cells {
    adder = std_add(32);
    outpt = std_reg(32);
  }
  wires {
    group do_add {
      adder.left = left;
      adder.right = right;
      outpt.in = adder.out;
      outpt.write_en = 1'd1;
      do_add[done] = outpt.done;
    }
  }
  control {
    seq {
      do_add;
    }
  }
}

component main() -> () {
  cells {
    my_add = add();
    y = std_reg(32);
  }
  wires {
    group wr_y {
      y.in = 32'd10;
      y.write_en = 1'd1;
      wr_y[done] = y.done;
    }
  }
  control {
    seq {
      wr_y;
      invoke my_add(left = y.out, right = y.out)();
    }
  }
}
```
removes `my_add` from the cells of main.

rachitnigam, 2021-03-24:
`dead-cell-removal`, like every other optimization, assumes that `compile-invoke` has been run. Related to #355.

sgpthomas, 2021-03-24:
does this remove the need for #355 then? if so we should close that issue and have the code fail instead of `unimplemented!()`

sgpthomas, 2021-03-24:
if that's true, we should also have `dead-cell-removal` fail if there is an `invoke` in the control

rachitnigam, 2021-03-24:
I don't think so. I think we should support `invoke` reasoning in all passes. I'm just saying that we should probably group this with #355.

sgpthomas, 2021-03-24:
ah I see. discovered this while working on #355 so I agree

commit bc3e73e1 (2021-04-01): Make relevant passes reason about invoke (#458)

* make dead cell removal reason about invoke (fixes #448)

* add test for minimize reg invoke

* correct expect files for invoke tests

* allow live range analysis, schedule_conflicts, and sharing_components to reason
about invoke

* make clippy happy

* perform optimizations before compile-invoke

* remove sigs from `ShareComponents::cell_filter` + added test for shareable user
defined components

* actually make things compile + remove print statements

* allow resource sharing to share user defined components
closed 2021-04-01 by sgpthomas