calyxir/calyx #457: No way to access non-primitive component signatures in visitor

sgpthomas, 2021-03-25:
In the visitor functions that you override to implement a pass, you are given a mutable reference to the pass data structure, the control ast node, the component, and a list of library signatures. Notably, there is no way to get the signatures of neighboring components (because the library signatures are just primitive signatures and not component signatures). This, for example, makes it impossible to extend resource sharing to work with user defined components because the pass can't tell if 
```
cells {
  ...
  my_add = custom_shareable_add(32);
  ...
}
```
`custom_shareable_add` is shareable. 

I _think_ that this is a remnant from when we had a stronger distinction between primitives and components. The solution is to extend `ir::LibrarySignatures` to hold all component signatures.

I suppose this complicates modifying component signatures. On changes, we would have to make sure to update the `Signatures` data structure that we give to passes. Maybe the solution is to make signature modifications more structured in some way and not allow passes to arbitrarily mutate signatures. @rachitnigam thoughts?

Another solution may be to give components a weak pointer back to the context which would allow passes to traverse backwards through the component hierarchy to find neighboring components.

rachitnigam, 2021-03-25:
Infer-static-timing reasons about attributes on user-level components. Worth summarising how it does it.

sgpthomas, 2021-03-31:
Can you point me to where this happens? I'm not finding it

sgpthomas, 2021-03-31:
never mind. I figured out how to do it, but I don't think `infer-static-timing` does it correctly. It should search through `&ctx.components` as well as `&ctx.libs.sigs.values()`

closed 2021-03-31 by sgpthomas