calyxir/calyx #459: Resource sharing performance regression

sgpthomas, 2021-03-25:
With the merging of minimize-regs and resource-sharing, resource sharing now takes way longer than it used to.

rachitnigam, 2021-03-26:
Can you explain why the generated graphs are much bigger? Or rather, how do they differ from the origin `resource-sharing` graphs?

sgpthomas, 2021-03-30:
previously, `resource-sharing` didn't built a graph of cells. It only built the graph of groups. It performed coloring directly on the graph of groups and only ever considered sharing cells of the same type. The new implementation builds a single graph with all the cells, uses the group conflict graph to add conflict edges, and then performs coloring. However, because cells of different types show up in the graph, conflict edges have to be added between cells of different types. For resource sharing this leads to large graphs with huge numbers of edges, because we have many shareable primitives. The same problem doesn't manifest with registers because the programs we've been testing don't use large numbers of different kinds of registers.

closed 2021-03-31 by sgpthomas