calyxir/calyx #465: P4 and Calyx brainstorming.

cgyurgyik, 2021-03-27:
As discussed in Slack, I'm looking to forge ahead with a small project as part of the CS 5114 curriculum. This issue is created to brainstorm different ideas that are both applicable and feasible.

Suggestions:
1. Dig into [petr4](https://github.com/cornell-netlab/petr4). - @rachitnigam 
2. For P4, perhaps tackle a piece of the pipeline. - @sampsyo

sampsyo, 2021-03-28:
Yo! This could be fun. Here are some assorted/disconnected thoughts…

- A main practical impediment may be just hooking the generated hardware up to do something interesting and networky. That is, if you want to do a "hello world" demo of something compiled from P4 to Calyx, for instance, you will want it to process packets—so where do those packets come from?
	- The maximalist answer is that we want to run the thing on a real FPGA, in which case there is a substantial amount of work to do just to interface with the physical Ethernet port and the host networking system. That's a lot of scaffolding! So one possible project would just be spinning up that scaffolding so even a simple, no-op P4 program can work. One option for this could be to build on/hijack the scaffolding from [SDNet](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2017_1/UG1012-sdnet-packet-processor.pdf).
	- A more minimalist answer is that we'll do stuff in simulation (Verilator). But even so, simulating packet processing will be different from simulating the kind of one-off accelerators we've built so far. In particular, you'll probably want to process more than one packet—you'll want to be able to read from some sort of recorded traffic dump, simulated the Calyx-generated hardware on every packet therein, and produce a new traffic dump as output. It might be worth checking out how the Petr4 interpreter works when it's just executing "for correctness," in some manner of testing mode: does it read and write traffic-dump files, or does it somehow hook into actual kernel sockets and process data "live"?
- To expand slightly on what I said on Slack, if the focus is to be on actually generating Calyx code from P4, I think it would probably be necessary to pick one _piece_ of P4 to implement. As I understand it, a typical P4 pipeline in hardware has three main pieces: the parser, the processing (consisting of match/action table applications), and the deparser (packet emitter). Maybe picking one of these three would be a good way to narrow stuff down.

EclecticGriffin, 2021-03-28:
I can add some more confusing nuance to the above in that the processing can be (roughly) divided into two stages, ingress and egress, with the Traffic Manager (crossbar) in-between responsible for queuing and routing packets from the input pipelines to the output pipelines. The TM isn't (currently) programmable via p4 and is usually viewed as an opaque component.

EclecticGriffin, 2021-03-28:
The match action components of the processing pipeline also pose an interesting challenge in that while the tables and actions are declared and written in the p4 code, the actual match rules associating match forms and subsequent actions are given by the control plane and subsequently are neither fixed by the program nor necessarily static over time.

sampsyo, 2021-03-28:
Good point, @EclecticGriffin. Maybe even figuring out an efficient TCAM primitive for Calyx would be a project.

jnfoster, 2021-03-29:
Can confirm there is *lots* of secret sauce in how TCAMs are implemented in industry (e.g., search for "cuckoo hash")

rachitnigam, 2021-03-30:
@jnfoster any thoughts on which part of this could be interesting and reasonably self-contained? I see at least three interesting things to design:

1. A "frontend" that transforms a prefix-based table into a fast, hardware based cuckoo hash table.
2. A simple, table-free fragment of Petr4 or P4 and turning it into hardware. 
3. A fast streaming implementation of a ingress packet parser.

cgyurgyik, 2021-03-30:
Thanks all, truly! This is more than I expected. I'll take some time to carefully read your responses, and we'll go from there.

cgyurgyik, 2021-04-01:
I've created a simple example below of how I understand the process. Please check my understanding.

## Setting
 Let's say we have the following table.
| Address | Action      |
|---------|-------------|
| 10.0.1  | GOTO_PORT11 |
| 10.0.2  | GOTO_PORT22 |
| 10.0.3  | GOTO_PORT33 |

For simplicity, the table consists of only exact matches, so that a CAM can be used.

From this table, we want to generate two pieces: 
### 1. The Match Engine. 
- Given data (an Address), returns an address index into the action memory. 
- Implemented as a CAM here (eventually, a TCAM for longest prefix matching (LPM).
- We'll assume here that the data passed in is the final digit of the Address, e.g. for `10.0.1`, this is `1`, or `001`.

| Data | Address |
|------|---------|
| 001  | 01      |
| 010  | 02      |
| 100  | 03      |

### 2. The Action Memory. 
- Given an address index, returns the corresponding action.
- Implemented as SRAM.
- Here, I've made the Action something human-readable, but this will instead be some N-bit code which will be eventually interpreted by an Action ALU.

| Address | Action      |
|---------|-------------|
| 01      | GOTO_PORT11 |
| 02      | GOTO_PORT22 |
| 03      | GOTO_PORT33 |

## Example
Let's say the address we receive is `10.0.2`.
1. In the Match Engine phase, Lookup(`10.0.2`) = Lookup(`2`) = Lookup(`010`) = `02`.
2. In the Action Memory phase, Lookup(`02`) = `GOTO_PORT22`.
3. Interpret with some Action ALU accordingly.

sampsyo, 2021-04-01:
This seems perfectly reasonable to me! I wouldn't have originally thought of the idea to split up the table so that the CAM produces an address into a standard SRAM for the second-stage lookup, but I can see how that keeps the CAM compact.

cgyurgyik, 2021-04-01:
> This seems perfectly reasonable to me! I wouldn't have originally thought of the idea to split up the table so that the CAM produces an address into a standard SRAM for the second-stage lookup, but I can see how that keeps the CAM compact.

Eventually I foresee the MatchEngine being a TCAM, so that it can use LPM efficiently. From some reading, it seems to either be implemented using Cucko hashing (direct match) or TCAM (LPM). I just used a CAM here to keep it simple and that's what I'll probably implement first.

closed 2021-05-10 by cgyurgyik