calyxir/calyx #494: Testing the Relay frontend against a developed infrastructure.

cgyurgyik, 2021-04-27:
We now have each function of the VGG Net implemented in Dahlia, at least to simulate a single example all the way through. While there exist a single test or two for each function individually, it brings into question the best way to test larger modules  and even entire nets.

For context, below is the Relay IR for a "Test VGG Net".

<details>
<summary>VGG Net Relay IR</summary>
<br>

```
fn (%data: Tensor[(5, 3, 224, 224), int32], %conv1_1_weight: Tensor[(64, 3, 3, 3), int32], %conv1_1_bias: Tensor[(64), int32], %bn1_1_gamma: Tensor[(64), int32], %bn1_1_beta: Tensor[(64), int32], %bn1_1_moving_mean: Tensor[(64), int32], %bn1_1_moving_var: Tensor[(64), int32], %conv1_2_weight: Tensor[(64, 64, 3, 3), int32], %conv1_2_bias: Tensor[(64), int32], %bn1_2_gamma: Tensor[(64), int32], %bn1_2_beta: Tensor[(64), int32], %bn1_2_moving_mean: Tensor[(64), int32], %bn1_2_moving_var: Tensor[(64), int32], %conv2_1_weight: Tensor[(128, 64, 3, 3), int32], %conv2_1_bias: Tensor[(128), int32], %bn2_1_gamma: Tensor[(128), int32], %bn2_1_beta: Tensor[(128), int32], %bn2_1_moving_mean: Tensor[(128), int32], %bn2_1_moving_var: Tensor[(128), int32], %conv2_2_weight: Tensor[(128, 128, 3, 3), int32], %conv2_2_bias: Tensor[(128), int32], %bn2_2_gamma: Tensor[(128), int32], %bn2_2_beta: Tensor[(128), int32], %bn2_2_moving_mean: Tensor[(128), int32], %bn2_2_moving_var: Tensor[(128), int32], %conv3_1_weight: Tensor[(256, 128, 3, 3), int32], %conv3_1_bias: Tensor[(256), int32], %bn3_1_gamma: Tensor[(256), int32], %bn3_1_beta: Tensor[(256), int32], %bn3_1_moving_mean: Tensor[(256), int32], %bn3_1_moving_var: Tensor[(256), int32], %conv3_2_weight: Tensor[(256, 256, 3, 3), int32], %conv3_2_bias: Tensor[(256), int32], %bn3_2_gamma: Tensor[(256), int32], %bn3_2_beta: Tensor[(256), int32], %bn3_2_moving_mean: Tensor[(256), int32], %bn3_2_moving_var: Tensor[(256), int32], %conv4_1_weight: Tensor[(512, 256, 3, 3), int32], %conv4_1_bias: Tensor[(512), int32], %bn4_1_gamma: Tensor[(512), int32], %bn4_1_beta: Tensor[(512), int32], %bn4_1_moving_mean: Tensor[(512), int32], %bn4_1_moving_var: Tensor[(512), int32], %conv4_2_weight: Tensor[(512, 512, 3, 3), int32], %conv4_2_bias: Tensor[(512), int32], %bn4_2_gamma: Tensor[(512), int32], %bn4_2_beta: Tensor[(512), int32], %bn4_2_moving_mean: Tensor[(512), int32], %bn4_2_moving_var: Tensor[(512), int32], %conv5_1_weight: Tensor[(512, 512, 3, 3), int32], %conv5_1_bias: Tensor[(512), int32], %bn5_1_gamma: Tensor[(512), int32], %bn5_1_beta: Tensor[(512), int32], %bn5_1_moving_mean: Tensor[(512), int32], %bn5_1_moving_var: Tensor[(512), int32], %conv5_2_weight: Tensor[(512, 512, 3, 3), int32], %conv5_2_bias: Tensor[(512), int32], %bn5_2_gamma: Tensor[(512), int32], %bn5_2_beta: Tensor[(512), int32], %bn5_2_moving_mean: Tensor[(512), int32], %bn5_2_moving_var: Tensor[(512), int32], %fc6_weight: Tensor[(4096, 25088), int32], %fc6_bias: Tensor[(4096), int32], %fc7_weight: Tensor[(4096, 4096), int32], %fc7_bias: Tensor[(4096), int32], %fc8_weight: Tensor[(10, 4096), int32], %fc8_bias: Tensor[(10), int32]) -> Tensor[(5, 10), int32] {
  let %x: Tensor[(5, 64, 224, 224), int32] = nn.conv2d(%data, %conv1_1_weight, padding=[1, 1, 1, 1], channels=64, kernel_size=[3, 3]) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x1: Tensor[(5, 64, 224, 224), int32] = nn.bias_add(%x, %conv1_1_bias) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x2: int32 = 1 /* ty=int32 */;
  let %x3: int32 = 0 /* ty=int32 */;
  let %x4: Tensor[(64), int32] = add(%bn1_1_moving_var, %x3) /* ty=Tensor[(64), int32] */;
  let %x5: Tensor[(64), int32] = sqrt(%x4) /* ty=Tensor[(64), int32] */;
  let %x6: Tensor[(64), int32] = divide(%x2, %x5) /* ty=Tensor[(64), int32] */;
  let %x7: Tensor[(64), int32] = multiply(%x6, %bn1_1_gamma) /* ty=Tensor[(64), int32] */;
  let %x8: Tensor[(64, 1, 1), int32] = expand_dims(%x7, axis=1, num_newaxis=2) /* ty=Tensor[(64, 1, 1), int32] */;
  let %x9: Tensor[(5, 64, 224, 224), int32] = multiply(%x1, %x8) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x10: Tensor[(64), int32] = negative(%bn1_1_moving_mean) /* ty=Tensor[(64), int32] */;
  let %x11: Tensor[(64), int32] = multiply(%x10, %x7) /* ty=Tensor[(64), int32] */;
  let %x12: Tensor[(64), int32] = add(%x11, %bn1_1_beta) /* ty=Tensor[(64), int32] */;
  let %x13: Tensor[(64, 1, 1), int32] = expand_dims(%x12, axis=1, num_newaxis=2) /* ty=Tensor[(64, 1, 1), int32] */;
  let %x14: Tensor[(5, 64, 224, 224), int32] = add(%x9, %x13) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x15: Tensor[(5, 64, 224, 224), int32] = nn.relu(%x14) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x16: Tensor[(5, 64, 224, 224), int32] = nn.conv2d(%x15, %conv1_2_weight, padding=[1, 1, 1, 1], channels=64, kernel_size=[3, 3]) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x17: Tensor[(5, 64, 224, 224), int32] = nn.bias_add(%x16, %conv1_2_bias) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x18: int32 = 1 /* ty=int32 */;
  let %x19: int32 = 0 /* ty=int32 */;
  let %x20: Tensor[(64), int32] = add(%bn1_2_moving_var, %x19) /* ty=Tensor[(64), int32] */;
  let %x21: Tensor[(64), int32] = sqrt(%x20) /* ty=Tensor[(64), int32] */;
  let %x22: Tensor[(64), int32] = divide(%x18, %x21) /* ty=Tensor[(64), int32] */;
  let %x23: Tensor[(64), int32] = multiply(%x22, %bn1_2_gamma) /* ty=Tensor[(64), int32] */;
  let %x24: Tensor[(64, 1, 1), int32] = expand_dims(%x23, axis=1, num_newaxis=2) /* ty=Tensor[(64, 1, 1), int32] */;
  let %x25: Tensor[(5, 64, 224, 224), int32] = multiply(%x17, %x24) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x26: Tensor[(64), int32] = negative(%bn1_2_moving_mean) /* ty=Tensor[(64), int32] */;
  let %x27: Tensor[(64), int32] = multiply(%x26, %x23) /* ty=Tensor[(64), int32] */;
  let %x28: Tensor[(64), int32] = add(%x27, %bn1_2_beta) /* ty=Tensor[(64), int32] */;
  let %x29: Tensor[(64, 1, 1), int32] = expand_dims(%x28, axis=1, num_newaxis=2) /* ty=Tensor[(64, 1, 1), int32] */;
  let %x30: Tensor[(5, 64, 224, 224), int32] = add(%x25, %x29) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x31: Tensor[(5, 64, 224, 224), int32] = nn.relu(%x30) /* ty=Tensor[(5, 64, 224, 224), int32] */;
  let %x32: Tensor[(5, 64, 112, 112), int32] = nn.max_pool2d(%x31, pool_size=[2, 2], strides=[2, 2], padding=[0, 0, 0, 0]) /* ty=Tensor[(5, 64, 112, 112), int32] */;
  let %x33: Tensor[(5, 128, 112, 112), int32] = nn.conv2d(%x32, %conv2_1_weight, padding=[1, 1, 1, 1], channels=128, kernel_size=[3, 3]) /* ty=Tensor[(5, 128, 112, 112), int32] */;
  let %x34: Tensor[(5, 128, 112, 112), int32] = nn.bias_add(%x33, %conv2_1_bias) /* ty=Tensor[(5, 128, 112, 112), int32] */;
  let %x35: int32 = 1 /* ty=int32 */;
  let %x36: int32 = 0 /* ty=int32 */;
  let %x37: Tensor[(128), int32] = add(%bn2_1_moving_var, %x36) /* ty=Tensor[(128), int32] */;
  let %x38: Tensor[(128), int32] = sqrt(%x37) /* ty=Tensor[(128), int32] */;
  let %x39: Tensor[(128), int32] = divide(%x35, %x38) /* ty=Tensor[(128), int32] */;
  let %x40: Tensor[(128), int32] = multiply(%x39, %bn2_1_gamma) /* ty=Tensor[(128), int32] */;
  let %x41: Tensor[(128, 1, 1), int32] = expand_dims(%x40, axis=1, num_newaxis=2) /* ty=Tensor[(128, 1, 1), int32] */;
  let %x42: Tensor[(5, 128, 112, 112), int32] = multiply(%x34, %x41) /* ty=Tensor[(5, 128, 112, 112), int32] */;
  let %x43: Tensor[(128), int32] = negative(%bn2_1_moving_mean) /* ty=Tensor[(128), int32] */;
  let %x44: Tensor[(128), int32] = multiply(%x43, %x40) /* ty=Tensor[(128), int32] */;
  let %x45: Tensor[(128), int32] = add(%x44, %bn2_1_beta) /* ty=Tensor[(128), int32] */;
  let %x46: Tensor[(128, 1, 1), int32] = expand_dims(%x45, axis=1, num_newaxis=2) /* ty=Tensor[(128, 1, 1), int32] */;
  let %x47: Tensor[(5, 128, 112, 112), int32] = add(%x42, %x46) /* ty=Tensor[(5, 128, 112, 112), int32] */;
  let %x48: Tensor[(5, 128, 112, 112), int32] = nn.relu(%x47) /* ty=Tensor[(5, 128, 112, 112), int32] */;
  let %x49: Tensor[(5, 128, 112, 112), int32] = nn.conv2d(%x48, %conv2_2_weight, padding=[1, 1, 1, 1], channels=128, kernel_size=[3, 3]) /* ty=Tensor[(5, 128, 112, 112), int32] */;
  let %x50: Tensor[(5, 128, 112, 112), int32] = nn.bias_add(%x49, %conv2_2_bias) /* ty=Tensor[(5, 128, 112, 112), int32] */;
  let %x51: int32 = 1 /* ty=int32 */;
  let %x52: int32 = 0 /* ty=int32 */;
  let %x53: Tensor[(128), int32] = add(%bn2_2_moving_var, %x52) /* ty=Tensor[(128), int32] */;
  let %x54: Tensor[(128), int32] = sqrt(%x53) /* ty=Tensor[(128), int32] */;
[... cut ...]
  let %x173: Tensor[(5, 10), int32] = nn.bias_add(%x172, %fc8_bias, axis=-1) /* ty=Tensor[(5, 10), int32] */;
  let %x174: Tensor[(5, 10), int32] = nn.softmax(%x173) /* ty=Tensor[(5, 10), int32] */;
  %x174
}
```

</details>

I think it may be better to create some testing environment that, given a Relay function,
1. Emits the necessary Tensorflow and Calyx programs.
2. Creates the necessary data for each (randomized, perhaps).
3. Runs the Tensorflow program, and simulates the Calyx program.
4. Asserts that they are within some tolerance.

rachitnigam, 2021-04-27:
This plan certainly sounds like it'll be a worthwhile investment in the long run. The question for us is what do we want to accomplish in the next month. We can focus on getting VGG working first which will probably requiring building a manual harness that does exactly the thing above for just the VGG net or we can start working on this generic infrastructure now and get back to VGG when it's operational.

One tantalizing option is working on the infrastructure but really cutting corners to make VGG workable first. Once that is done, we can work on making the infrastructure more general purpose.

cgyurgyik, 2021-04-30:
Discussed today:
- Run directly with Relay rather than TF.
- Look for pre-calculated VGG net weights.
- Verify that VGG net lowers to Verilog

rachitnigam, 2021-04-30:
Looks like pretrained models can be found here: https://github.com/onnx/models

cgyurgyik, 2021-05-13:
Re-opening. I've provided a simple script that will output files for the Calyx program and run the TVM execution. We still need to bridge the gap of comparing the final output of each Softmax memory (and perhaps intermediary values as well).