calyxir/calyx #497: Bit splitting in Calyx.

cgyurgyik, 2021-05-01:
Sometimes we only want to access a certain subset of the bits in a multi-bit bus. `std_slice` provides limited functionality in this regard.

For example, in SV, we could access the `i`th bit of `in` with: `in[i]`.
Following this, a Calyx interface for a single-bit split primitive might look like:

```
module std_split #(
  parameter WIDTH,
  parameter INDEX_SIZE
) (
  input wire logic [WIDTH-1:0]      in,
  input wire logic [INDEX_SIZE-1:0] index,
  output logic out;
);
  assign out = in[index];

  `ifdef VERILATOR
    always_comb begin
      if (index < 0 || index > (WIDTH - 1))
        $error(
          "std_split: Index is outside the bounds of the WIDTH.\n",
          "WIDTH: %0d", WIDTH,
          "index: %0d", index
        );
    end
  `endif
endmodule
```

This would require N individual `std_split`s to access each bit in a N-bit bus. Since we're just "adding wires", perhaps there's a better approach for this?

For example, we could also add an `OUT_WIDTH` if we want to split off a subset of bits with size > 1. We would then also need to provide a  `begin_index` and `end_index`.

sampsyo, 2021-05-02:
Yes! It does seem like static vs. dynamic slices should probably be different primitives altogether. Static slices would get their indices as meta-level parameters, while dynamic slices would get their inputs as input wires. A good reason to separate them is that their costs are very different: static slices are effectively "free" in hardware, while dynamic slices are not.

rachitnigam, 2021-05-25:
Would it suffice to make `std_slice` take a `begin_index` and `end_index` instead of just the `out_index`? For reference, `std_slice` does this:

```verilog
  assign out = in[OUT_WIDTH-1:0];
```

anthonyabeo, 2024-01-20:
@rachitnigam I will like to tackle this issue. I suppose the plan is to go with your suggestion [here](https://github.com/calyxir/calyx/issues/497#issuecomment-847903821)?

sampsyo, 2024-01-22:
Hi, @anthonyabeo—I think that would be a good place to start! We would probably want to create a new, different primitive (to avoid conflicts with code that currently uses `std_slice`), and then consider migrating code over to it that could use the new primitive?

anthonyabeo, 2024-01-23:
@sampsyo sure, that sounds good. I'll put something together and revert.

closed 2024-07-22 by rachitnigam