calyxir/calyx #498: Writing to a register using `mult_pipe` with `<"static"=3>` never enables writes.

cgyurgyik, 2021-05-04:
<details>
<summary>Calyx Program</summary>
<br>

```
import "primitives/std.lib";
component main() -> () {
  cells {
    @external(1) slhs = std_mem_d1(4, 1, 1);
    @external(1) srhs = std_mem_d1(4, 1, 1);
    @external(1) signed_mul = std_mem_d1(4, 1, 1);
    smul = std_fp_smult_pipe(4,2,2);
    t = std_reg(4);
  }
  wires {
    group do_mul {
      slhs.addr0 = 1'd0;
      srhs.addr0 = 1'd0;
      smul.left = slhs.read_data;
      smul.right = srhs.read_data;
      t.in = smul.out;
      smul.go = !smul.done ? 1'd1;
      t.write_en = smul.done;
      do_mul[done] = t.done ? 1'd1;
    }
    group save_index {
      signed_mul.addr0 = 1'd0;
      signed_mul.write_data = t.out;
      signed_mul.write_en = 1'd1;
      save_index[done] = signed_mul.done;
    }
  }
  control {
    seq { do_mul; save_index; }
  }
}
```

</details>

<details>
<summary>Data</summary>
<br>

```
{
  "slhs": {
    "data": [
      -0.5
    ],
    "format": {
      "numeric_type": "fixed_point",
      "is_signed": true,
      "width": 4,
      "frac_width": 2
    }
  },
  "srhs": {
    "data": [
      1.0
    ],
    "format": {
      "numeric_type": "fixed_point",
      "is_signed": true,
      "width": 4,
      "frac_width": 2
    }
  },
  "signed_mul": {
    "data": [
      0.0
    ],
    "format": {
      "numeric_type": "fixed_point",
      "is_signed": true,
      "width": 4,
      "frac_width": 2
    }
  }
}
```

</details>

If we lower this using `fud e test.futil --to futil-lowered`:
```
import "primitives/std.lib";
component main<"static"=4>(go: 1, clk: 1) -> (done: 1) {
  cells {
    @external(1) slhs = std_mem_d1(4, 1, 1);
    @external(1) srhs = std_mem_d1(4, 1, 1);
    @external(1) signed_mul = std_mem_d1(4, 1, 1);
    smul = std_fp_smult_pipe(4, 2, 2);
    t = std_reg(4);
    fsm = std_reg(3);
    incr = std_add(3);
  }
  wires {
    done = fsm.out == 3'd4 ? 1'd1;
    fsm.clk = clk;
    fsm.in = fsm.out == 3'd4 ? 3'd0;
    fsm.in = fsm.out != 3'd4 & go ? incr.out;
    fsm.write_en = fsm.out != 3'd4 & go | fsm.out == 3'd4 ? 1'd1;
    incr.left = go ? 3'd1;
    incr.right = go ? fsm.out;
    signed_mul.addr0 = fsm.out == 3'd3 & go ? 1'd0;
    signed_mul.clk = clk;
    signed_mul.write_data = fsm.out == 3'd3 & go ? t.out;
    signed_mul.write_en = fsm.out == 3'd3 & go ? 1'd1;
    slhs.addr0 = fsm.out < 3'd3 & go ? 1'd0;
    slhs.clk = clk;
    smul.clk = clk;
    smul.go = !smul.done & fsm.out < 3'd3 & go ? 1'd1;
    smul.left = fsm.out < 3'd3 & go ? slhs.read_data;
    smul.right = fsm.out < 3'd3 & go ? srhs.read_data;
    srhs.addr0 = fsm.out < 3'd3 & go ? 1'd0;
    srhs.clk = clk;
    t.clk = clk;
    t.in = fsm.out < 3'd3 & go ? smul.out;
    t.write_en = fsm.out < 3'd3 & go ? smul.done;
  }

  control {}
}
```

- offending line: `t.write_en = fsm.out < 3'd3 & go ? smul.done;`

If we look at the waveform, we'll see this never occurs. Wires displayed (from bottom -> up):
```
t.write_en
smul.clk
smul.done
smul.go
fsm.out
```

[image]

rachitnigam, 2021-05-04:
Shouldn't it be `static=4` for the group since the multiplier alone takes 3 cycles.

cgyurgyik, 2021-05-04:
Ugh yeah I think that's it. Probably a bug related to my Dahlia PR.

closed 2021-05-04 by cgyurgyik