calyxir/calyx #505: `exp` operator producing negative values when `x` is a large negative number.

cgyurgyik, 2021-05-08:
Program:
```
python3 calyx-py/calyx/gen_exp.py tests/correctness/exp/degree-4-signed.txt
```
Data:
```
{
  "x": {
    "data": [
      -4000.0
    ],
    "format": {
      "numeric_type": "fixed_point",
      "is_signed": true,
      "width": 32,
      "int_width": 16
    }
  },
  "ret": {
    "data": [
      0.0
    ],
   "format": {
      "numeric_type": "fixed_point",
      "is_signed": true,
      "width": 32,
      "int_width": 16
    }
  }
}
```

Output: `"-0.00189208984375"`
This should be `0`.

cgyurgyik, 2021-05-09:
This may be a good starting point to transition to an actual LUT `exp` rather than my Taylor Series expansion.

rachitnigam, 2021-11-23:
While I am able to simulate the component and reproduce the error, I have no clue why it happens. The interpreter says there is an overflow:
```
WARN - Over/underflow in fixed-point multiplier: 69218.52006149594672024250030517578125 to 3682.520050048828125
WARN - Over/underflow in fixed-point multiplier: 73964.9963769479654729366302490234375 to 8428.996368408203125
```
So maybe we have a place to root-cause this.

To reproduce, run `fud e --to interpreter-out exp.futil -s verilog.data exp.json -vv`

cgyurgyik, 2021-11-23:
When the type is signed and the value is less than zero, the `exp` generator will compute `e^-4000` as `1 / e^4000`:
https://github.com/cucapra/calyx/blob/a14ce9d743d882d3cadd927cbb3be797a3dc42bf/calyx-py/calyx/gen_exp.py#L503-L536

Since this has no decimal places in it, we are computing `e^4000` as `fp_pow(e, 4000)`:

https://github.com/cucapra/calyx/blob/a14ce9d743d882d3cadd927cbb3be797a3dc42bf/calyx-py/calyx/gen_exp.py#L632-L643

A simple fix is to add a cut-off if `x` is a very large negative number, and instead return `0`.

rachitnigam, 2021-11-24:
Thanks for explaining the problem! One possible option is implementing “saturating” mathematical operators which, instead of overflowing, clamp the outputs to either a maximum or a minimum. So In this case, it’d clamp the value generated by the divider to 0.

rachitnigam, 2021-11-27:
For anyone who wants to work on this issue, there are two possible options:
1. Update the exp generator to have a check for extremely large negative numbers and just return `0` when it meets some threshold.
2. Implement a ["saturating divide"](https://en.wikipedia.org/wiki/Saturation_arithmetic) operator in the primitive library and use that instead of the standard divider. This might come down to implementing the threshold check inside the divider instead of the exp.