calyxir/calyx #508: Making the Relay call node implementations maintainable.

cgyurgyik, 2021-05-13:
A general (and unhelpful) comment, but this is a good example of what I'm thinking so I'll leave it here. 

I get nervous when we have situations where there are load-bearing strings and am wondering if at some point it would be possible to refactor into a nice set of constructors. Though that would likely require having either an internal representation of Dahlia or a Dahlia python library. Obviously this isn't actionable in the short term nor in the scope of this PR, so please don't worry about it.

_Originally posted by @EclecticGriffin in https://github.com/cucapra/calyx/pull/504#discussion_r631128511_

cgyurgyik, 2021-05-13:
Response from #504:
> Same page. The entire `dahlia_impl.py` file hurts my eyes; unfortunately I am also the one to blame. 
> For context, when we first started this in the fall, our goal was to simulate a net using Calyx. In an ideal world, all of our Relay call nodes would be implemented in Verilog or Calyx... hopefully one day that can actually be the case. For now, we decided using Dahlia to implement the function and then lower it to Calyx would be a more efficient process.
>
> While outside the scope of this review, the next question is what are the steps to make this easier to maintain and read. Possibilities:
>
> 1. Python AST for Dahlia. Perhaps a good intermediary step, but still keeps Dahlia in the picture (part of the reason I've been hesitant to make any significant changes to this file for now).
> 2. Use Python scripts to generate the correct Calyx component for each Relay call node (similar to the [`exp` generator](https://github.com/cucapra/calyx/blob/master/calyx-py/calyx/gen_exp.py)). Adrian mentions a cleaner way to do this [here](https://github.com/cucapra/calyx/discussions/419).
> 3. SystemVerilog implementations of the Relay call nodes.

cgyurgyik, 2021-05-13:
File in question: https://github.com/cucapra/calyx/blob/master/frontends/relay/dahlia_impl.py

sampsyo, 2021-05-13:
I can't resist mentioning #419 and the vague metaprogramming ideas contained therein. This Dahlia generation infrastructure could be a prime use case for driving that kind of thinking.

cgyurgyik, 2021-05-13:
> Adrian mentions a cleaner way to do this [here](https://github.com/cucapra/calyx/discussions/419).

Yep! I didn't realize that Discussions take a # as well.

rachitnigam, 2021-09-24:
We've decided to make this a future project for when someone can take ownership of the Relay frontend. We're turning this issue into a project idea in the projects list.

closed 2021-09-24 by rachitnigam