calyxir/calyx #509: Interpreter not updating when done signal of the group is directly assigned

yoonachang, 2021-05-13:
The interpreter does not run the updates in the group if the done signal is assigned true. Thus, the done signal of the group needs be given indirectly with a register in the current state. This needs to be fixed so that the done signal of the group gets assigned 1 after all the updates are done even it is directly given  1'd1.

A code that compiles but does not work with the current interpreter
```
  wires {
    group cond<"static"=0> {
      lt0.left = const0.out;
      lt0.right = const1.out;
      cond[done] =  1'd1;
    }
```
A code that compiles and works with the current interprete```
  wires {
    group cond<"static"=0> {
      reg0.write_en = 1'd1;
      lt0.left = const0.out;
      lt0.right = const1.out;
      reg0.in = lt0.out;
      cond[done] = reg0.done;
    }```

rachitnigam, 2021-07-05:
@EclecticGriffin does the new interpreter have this issue/similar issue?

EclecticGriffin, 2021-07-05:
I'm pretty sure the current iteration of the interpreter does not suffer this issue

EclecticGriffin, 2021-08-16:
Closing since this issue is no longer present/relevant

closed 2021-08-16 by EclecticGriffin