calyxir/calyx #510: Metadata for ir structures

EclecticGriffin, 2021-05-14:
This originally came up in the course of implementing register unsharing (#417) but warrants its own discussion. 

## The Problem
Passes, such as the reaching definition analysis, may want or need to attach extra metadata to specific IR structures. In the case of unsharing, the analysis pass needs to attach transient labels to Invoke statements which are consumed during the re-writing process. This type of metadata is likely to be useful in other passes, so considering a general mechanism may be warranted.

### Why not Attributes?
It is possible for attributes to fill this roll as passes can add to them, however as attributes are user facing extra care must be taken to avoid naming conflicts and they must be removed after the pass to ensure transience. There is an additional expressiveness problem in that attributes currently only map strings to a u64 which might not be sufficient depending on what type of metadata is desired. 

Additionally, altering the attributes requires a mutable reference to the ir struct and we would like analyses to be able to use metadata without needing the ability to alter the IR itself.

## Requirements
* Attach metadata to individual IR constructs
* Either a clear understanding of what type metadata is or a means of keeping it flexible
* Transience. Metadata should be strictly internal and ideally not leak between different analyses

## Preliminary Approach Ideas

### Lookup table
Analyses can create an external table that can be used to map ir structures to the corresponding metadata. This has the virtue of enabling flexibility on the metadata type and makes it transient by default as the pass wishing to use the analysis will have to consume the metadata object.

Being able to index the table by an arbitrary ir struct requires some consideration. Possible options include

#### Create a UUID for each struct
This is conceptually simple but may be challenging to implement as it requires a means of generating unique identifiers for each construct during its creation.

#### Use each construct's raw pointer as the key
This does not require changes to struct definitions or constructors and does not require unsafe, provided we don't attempt to dereference the keys and only use the keys as a way into the table. This does risk leaving dangling pointers in the key set, however, these will vanish when the metadata table is dropped and should be safe, provided we limit the ability to extract keys from the table itself. 

### Attach a metadata table directly to the IR struct
This approach makes attachment very easy but requires more care to avoid metadata spilling between analyses (the seriousness of which is dependent on how much we trust ourselves) and will place limits on what types metadata can be as it will require a type definition within the ir constructs. In all likelihood this will force us to use strings as the values which may make passes that consume metadata have to do extra work to parse the information.

sampsyo, 2021-05-15:
Great overview! I wonder if we can come up with other use cases for this "extra data structures on the side" pattern… doing so could help choose the right solution.

EclecticGriffin, 2021-05-17:
I'll see if I can come up with some other examples. My intuition as to why this hasn't come up yet is because a lot of the existing analyses are concerned with manipulating groups and cells, both of which have identifiers in the form of names. The reason metadata came up for register unsharing was the need to manipulate the control structure itself in a targeted way. I wonder if there are loop based analyses that might be portable to this setting. Seems a daunting prospect.

sampsyo, 2021-09-24:
We don't currently have a need for this, so deferring it to when we actually do. One thing we could do later is try to copy whatever LLVM does with invalidation of analysis pass results.

closed 2021-09-24 by sampsyo