calyxir/calyx #514: Replace `groups` and `cells` Vecs in Component with LinkedHashMaps

EclecticGriffin, 2021-05-21:
Replacing these will allow the `find_cell` and `find_group` methods to operate in constant time rather than linear time, but will preserve iteration order. The change is not likely to be complicated but will require a bunch of places to adjust to the new signature. It may be helpful to create a wrapper method on the component to allow mut/imut iteration over these fields, rather than exposing them directly, though if Ids are changed for some reason there is the danger that the key will become out of sync with the value. This perhaps deserves a little consideration, though it would be relatively straightforward, albeit in linear time, to update the keys to match the values. 

See following thread for initial info:

Can you talk a bit more about what this is amortizing exactly? Is it because the `find_cell` and `find_group` methods are linear? The right way to fix that is just to make them use a `LinearHashMap` and have constant access in those methods directly.

_Originally posted by @rachitnigam in https://github.com/cucapra/calyx/pull/511#discussion_r634451397_

EclecticGriffin, 2021-05-24:
Upon consideration and discussion with Sam, it may make more sense to try to use a BTreeSet and define ordering & hashing over the RRCs based on their id. The main issue is that if we have the keys into a map be the ids of the corresponding group or cell then they could become out of sync with the value if someone changes the value's id.

EclecticGriffin, 2021-05-24:
Should also wrap over this with an interface so that the particular implementation choices don't matter

rachitnigam, 2021-05-25:
> The main issue is that if we have the keys into a map be the ids of the corresponding group or cell then they could become out of sync with the value if someone changes the value's id.

One way out of this is making the underlying `id` of groups and cells be a `pub(super)` to the `ir::` crate and expose a function called `id()` which only gives an immutable handle to the id, disallowing any changes to it.

> Should also wrap over this with an interface so that the particular implementation choices don't matter

Aren't `find_cell` and `find_group` that two interface methods that do this already?

EclecticGriffin, 2021-05-25:
I think the `id()` idea would likely solve the main issue here. wrt to the second point, I meant iteration over the groups and cells through the component as well as replacing direct invocation of `push` on the underlying vector with some method on the component itself

rachitnigam, 2021-05-25:
Got it. You're right—we should abstract those patterns.

PR #516 (2021-06-07): Find fix
Closes #514. 

This was a painful process for what is ultimately a largely trivial change. Most of the changes aren't that interesting, as I had to correct every direct access to groups, cells, and the names thereof. However, I did have to add a little code here and there to address issues which came up due to subtle changes in mutability when directly accessing component fields. 

Also for annoying reasons, I had to make a few things closures which were previously iterators.

Anyway, making this a draft PR for the moment b/c I want to get your takes on this.
closed 2021-06-07 by EclecticGriffin