calyxir/calyx #520: Bug in if compliation for non-combinational conditional groups

EclecticGriffin, 2021-06-03:
The following program produces incorrect results. It erroneously takes the false branch.

```
import "primitives/core.futil";

component main() -> () {
  cells {
    mem = std_mem_d1(32, 1, 1);
    lt = std_lt(32);
    reg0 = std_reg(1);
  }

  wires {
    group cond<"static"=1> {
      reg0.write_en = 1'd1;
      lt.left = 32'd9;
      lt.right = 32'd16;
      reg0.in = lt.out;
      cond[done] = reg0.done;
    }

    group true<"static"=1> {
      mem.write_en = 1'd1;
      mem.addr0 = 1'd0;
      mem.write_data = 32'd4;
      true[done] = mem.done;
    }

    group false<"static"=1> {
      mem.write_en = 1'd1;
      mem.addr0 = 1'd0;
      mem.write_data = 32'd10;
      false[done] = mem.done;
    }
  }

  control {
    if lt.out with cond {
      true;
    } else {
      false;
    }
  }
}
```

When lowered we get 

```
import "primitives/core.futil";
component main<"static"=3>(go: 1, clk: 1) -> (done: 1) {
  cells {
    mem = std_mem_d1(32, 1, 1);
    lt = std_lt(32);
    reg0 = std_reg(1);
    fsm = std_reg(2);
    cond_stored = std_reg(1);
    incr = std_add(2);
  }
  wires {
    done = fsm.out == 2'd3 ? 1'd1;
    cond_stored.clk = clk;
    cond_stored.in = fsm.out == 2'd1 & go ? lt.out;
    cond_stored.write_en = fsm.out == 2'd1 & go ? 1'd1;
    fsm.clk = clk;
    fsm.in = fsm.out == 2'd3 ? 2'd0;
    fsm.in = fsm.out != 2'd3 & go ? incr.out;
    fsm.write_en = fsm.out != 2'd3 & go | fsm.out == 2'd3 ? 1'd1;
    incr.left = go ? fsm.out;
    incr.right = go ? 2'd1;
    lt.left = fsm.out < 2'd1 & go ? 32'd9;
    lt.right = fsm.out < 2'd1 & go ? 32'd16;
    mem.addr0 = fsm.out > 2'd1 & fsm.out < 2'd3 & cond_stored.out & go | fsm.out > 2'd1 & fsm.out < 2'd3 & !cond_stored.out & go ? 1'd0;
    mem.clk = clk;
    mem.write_data = fsm.out > 2'd1 & fsm.out < 2'd3 & !cond_stored.out & go ? 32'd10;
    mem.write_data = fsm.out > 2'd1 & fsm.out < 2'd3 & cond_stored.out & go ? 32'd4;
    mem.write_en = fsm.out > 2'd1 & fsm.out < 2'd3 & cond_stored.out & go | fsm.out > 2'd1 & fsm.out < 2'd3 & !cond_stored.out & go ? 1'd1;
    reg0.clk = clk;
    reg0.in = fsm.out < 2'd1 & go ? lt.out;
    reg0.write_en = fsm.out < 2'd1 & go ? 1'd1;
  }

  control {}
}
```

The `cond_stored` register is introduced to capture the conditional value; however it waits until fsm state 1 to take new values, by which point the value from `lt.out` is no longer available. It seems that in order to preserve desired behavior, the group's values need to remain high for a cycle after the group finishes, or the conditional value needs to be written in the same fsm state that allows values into `lt.right` and `lt.left` which would mean it would need to overlap with the execution of the group.

rachitnigam, 2021-06-04:
Also related to #405. This probably occurs before Dahlia only generates combinational groups for conditions. For a real fix, we should have separate style of combinational groups that are handled in a different compiler pass before compiling the control structures.

rachitnigam, 2021-06-07:
When running:
```
⇒  fud e --to dat cond-comp.futil -s verilog.data data.json
{
  "cycles": 10,
  "memories": {
    "mem": [
      10
    ]
  }
}
```

Disabling `static-timing` generates the right value:
```
⇒  fud e --to dat cond-comp.futil -s verilog.data data.json -s futil.flags "-d static-timing"
{
  "cycles": 12,
  "memories": {
    "mem": [
      4
    ]
  }
}
```

commit 2068dd55 (2021-08-02): Fix #520
commit 0cb608dc (2021-08-20): Pass to remove combinational groups (#620)

* check for comb groups in normal control flow

* Pass to remove combinational groups

* remove-comb-groups now works

* update interp test because of hard error

* enable remove-comb-groups for default compilation

* static-timing disallows combinational groups

* Fix #520

* Update tests

* update compile-empty to not generate combinational group

* update tests

* update cycle count

* empty tcam.sv not needed

* match_element does not use outside done signal

* fix tcam implementation

* run remove-comb-groups before infer-static-timing

* `remov [... cut]
closed 2021-08-20 by rachitnigam