calyxir/calyx #526: Register unsharing bug

rachitnigam, 2021-06-07:
The durbin benchmark in the polybench benchmarks generates incorrect values when optimized using register unsharing.

Dahlia program:
```
// BEGIN macro definitions

// END macro definitions

decl r_int: ubit<32>[8];
decl y_int: ubit<32>[8];

// XXX(rachit): This should be a local array.
decl z_int: ubit<32>[8];

let alpha: ubit<32> = 0 - r_int[0];
let beta: ubit<32> = 1;
y_int[0] := alpha;
---
for (let k: ubit<4> = 1..8) {
  beta := (1 - alpha * alpha) * beta;

  let sum: ubit<32> = 0;
  let i: ubit<4> = 0;
  while (i < k) {
    decor "#pragma HLS loop_tripcount min=1 max=7 avg=4"
    sum := sum + r_int[k-i-1] * y_int[i];
    // Update loop counter
    i := i + 1;
  }
  ---
  // modification to prevent div by 0
  if (beta != 0) {
    alpha := 0 - (r_int[k] + sum)/beta;
  } else {
    alpha := 0;
  }

  i := 0;
  while(i < k) {
    decor "#pragma HLS loop_tripcount min=1 max=7 avg=4"
    let y_i: ubit<32> = y_int[i];
    ---
    z_int[i] := y_i + alpha * y_int[k-i-1];
    // Update loop counter
    i := i + 1;
  }
  ---
  i := 0;
  while(i < k) {
    decor "#pragma HLS loop_tripcount min=1 max=7 avg=4"
    y_int[i] := z_int[i];
    // Update loop counter
    i := i + 1;
  }
  ---
  y_int[k] := alpha;
}
```

Data:
```
{
  "r_int0": {
    "data": [
      80,
      54,
      53,
      85,
      94,
      5,
      3,
      92
    ],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": false,
      "width": 32
    }
  },
  "y_int0": {
    "data": [
      67,
      73,
      51,
      45,
      8,
      76,
      24,
      69
    ],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": false,
      "width": 32
    }
  },
  "z_int0": {
    "data": [
      29,
      5,
      46,
      35,
      27,
      0,
      0,
      3
    ],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": false,
      "width": 32
    }
  }
}
```

Commands:
Without register unsharing:
```
fud e polybench/linear-algebra-durbin.fuse --to dat -o unopt.json -s verilog.data polybench/linear-algebra-durbin.fuse.data
```

With register sharing
```
fud e polybench/linear-algebra-durbin.fuse --to dat -o opt.json -s verilog.data polybench/linear-algebra-durbin.fuse.data -s futil.flags "-p validate -p infer-static-timing -p collapse-control -p register-unsharing -p compile-invoke -p compile -p post-opt -p lower"
```

rachitnigam, 2021-06-07:
Test minimization that still generates different outputs:
```
decl r_int: bit<32>[8];
decl y_int: bit<32>[8];
decl z_int: bit<32>[8];

let alpha: bit<32> = r_int[0];
y_int[0] := alpha;
---
for (let k: bit<4> = 1..8) {
  let sum: bit<32> = r_int[0] * y_int[0];
  ---
  alpha := (r_int[k] + sum) * alpha;
  z_int[0] := y_int[0];
  ---
  y_int[k] := alpha;
}
```

EclecticGriffin, 2021-06-08:
The error was (unsurprisingly) in the while. The analysis didn't re-incorporate the definitions flowing through the conditional when computing the fix-point meaning the analysis could separate definitions that shouldn't be separated because we can't assume a while kills a definition arriving from outside it b/c the while may not run.

closed 2021-06-08 by EclecticGriffin