calyxir/calyx #538: Multi-component simulation

sampsyo, 2021-06-14:
We should be able to simulate Calyx programs with more than one component! This issue is about the _structural_ version of multi-component programs (i.e., not using the `invoke` control operator). That is, you might have one component that "runs" a different component by explicitly assigning to its `go` port and then waiting for its `done` port to go from 0 to 1.

A challenge in this effort will be concurrency, i.e., running the "parent" component and the "child" component at the same time.

rachitnigam, 2021-06-16:
It’s worth noting at this point that we need to decide if go/done signals for components are a part of the actual semantics or not. With the invoke operator, it is not actually necessary to have go/done signals in the semantics of non structural programs.

sampsyo, 2021-06-16:
Indeed. I guess the default position is that they are required if you're running a component structurally but not for `invoke`. That's a little weird, TBH, but maybe a practical decision.

PR #661 (2021-09-16): Multicomponent simulation
This is a big one with a lot of changes to the internals of the interpreter which I will do my best to summarize below. Closes #539 and closes #538 (probably).

Environments now have an explicit lifetime annotation to the top-level borrowing context which keeps the borrows for the component defns and control structures alive. This is needed to support ComponentInterpreters as a Primitive which means that the interpreter trait also requires the lifetime bounds. There are a lot of lifetime annotations in here, unfortunately unavoidable.

Environments are now also attach [... cut]
closed 2021-09-16 by EclecticGriffin