calyxir/calyx #541: Fix up interp test suite

almathaler, 2021-06-14:
- [ ] We decided CLK port will always be zero for every component. So modify .expect files to expect this.
- [ ] Some tests (to be determined which) fail because improper syntax/compilation errors. Fixing up those tests.
- [ ] Some tests miss because .expect files don't exist. Compile + run these tests with verilator to see expected output, and write their .expect files.

rachitnigam, 2021-06-15:
>  We decided CLK port will always be zero for every component. So modify .expect files to expect this.

@cucapra/calyx-interpreter we can also disable the compiler pass that adds the `clk` signal to components. I assume this is specifically talking about primitive components? Again, we can simply choose to not have a `clk` in the primitive implementation. The downside is that when we try to link to Verilator, we have to create a shim `clk` etc. Not asking for a decision right now, just throwing out thoughts.

sampsyo, 2021-06-16:
Right… broadly, it is a little odd that we do add a clock to designs _in Calyx_ because Calyx semantics definitely don't have anything interesting to say about clock signals. But the trade-off, of course, is that, after inserting the clock, it's now very very easy to translate to working Verilog.

In the status quo, it's a nice goal to have the interpreter be capable of simulating _every_ Calyx program after _any_ pass runs, including the ones where the clock has been inserted. But @rachitnigam makes a good point that maybe this should be rolled into the Verilog backend instead of showing up in Calyx code.

In any case, I think forging ahead with constant-zero `clk` signals in the interpreter is a practical decision that will create no invalidated work in case we decide to remove `clk` from Calyx-land altogether someday.

EclecticGriffin, 2021-06-28:
I think we can close this now that #554 has been merged. Thoughts?

sampsyo, 2021-06-28:
Yeah, I think we've done everything in @almathaler's list above!

closed 2021-06-28 by almathaler