calyxir/calyx #547: [Interpreter] Correctly handle done signals for groups with arbitrary simulation time

EclecticGriffin, 2021-06-16:
Currently the done signal of registers and memories is returned as a time-locked value which is initially zero and later becomes one. Unfortunately this means that there is nothing to indicate that the signal should go low after another cycle. Done signals are moved to low when the group evaluation stops, however this still causes incorrect behavior with respect to done signals in the middle of a group execution, e.g. a sequence of register writes chained together. A good way to approach this might be the introduction of another `OutputValue` variant, possible called a pulse, which retains the fact that a high signal is to go low after a cycle. This could just be a wrapper over two time locked vals, but it might make more sense as its own thing.

Alternatively, there could be special logic in `do_tick` to search for high done signals and move them to low, though I find this less appealing and less maintainable.

EclecticGriffin, 2021-06-16:
This is done with the addition of `PulseValue` which is now used for the done output on stateful components. Additionally to work better with #543 , components no longer explicitly lower the done signal during normal execution, but do so during a special `reset` method

rachitnigam, 2021-06-17:
🔥 🔥 🔥

closed 2021-06-23 by EclecticGriffin