calyxir/calyx #553: Add More Primitives

almathaler, 2021-06-21:
We've added the core primitives but have yet to add those in `binary_operators/math.sv`. This work will lead into implementing the current primitives with binary arithmetic rather than casting values to and from u64, because the new primitives are fixed point.

EclecticGriffin, 2021-06-21:
just going to add a checklist here of the ops.

**Unsigned Fixed Point**

- [ ] `std_fp_add`
- [ ] `std_fp_sub`
- [ ] `std_fp_mult_pipe`
- [ ] `std_fp_div_pipe`
- [ ] `std_fp_gt`

**Signed Fixed Point**

- [ ] `std_fp_sadd`
- [ ] `std_fp_ssub`
- [ ] `std_fp_smult_pipe`
- [ ] `std_fp_sdiv_pipe`
- [ ] `std_fp_sgt`

**Unsigned Bitnum**

- [ ] `std_mult_pipe`
- [ ] `std_div_pipe`

**Signed Bitnum**

- [ ] `std_smult_pipe`
- [ ] `std_sdiv_pipe`
- [ ] `std_sgt`
- [ ] `std_slt`
- [ ] `std_seq`
- [ ] `std_sneq`
- [ ] `std_sge`
- [ ] `std_sle`
- [ ] `std_slsh`
- [ ] `std_srsh`

almathaler, 2021-07-08:
Update on `std_mult_pipe`: 
- Current `std_mult_pipe` isn't actually pipelined bitwise multiplication; still using the `u64` hack
- Even once we transition away from `u64` hack, the internals of `std_mult_pipe` probably won't take 3 cycles, as `std_mult_pipe` is designed to take in `calyx/primitives/binary_operators.sv`. So how do we coordinate the fact that the multiplier requires more than 1 cycle to commit its updates without the multiplier, being a primitive, having any knowledge of the clock or ticks?
- First attempt is to just have the update be a TLV, and each call to `commit_updates` decrement the count, finally committing the update once the TLV is unlockable.

sampsyo, 2021-07-08:
All sounds good, but just putting a note here to say that chatting more about plans for dealing with latency in primitives (and subcomponents in general) would be a good item for tomorrow's agenda.

EclecticGriffin, 2021-08-16:
Just marking updates on this, we have support for the three cycle unsigned pipeline mult and divide (added in #609, I believe). The rest are still open to my knoweldge

rachitnigam, 2021-10-13:
Should we close this in favor of an issue to deprecate `std_fp_add_dwidth` etc.? In a previous issue, we discussed removing those specialized primitives and using just the extend and slice primitives to get the right number of bits for the normal `std_fp_*` operators.

EclecticGriffin, 2021-10-13:
Sounds good to me

sampsyo, 2021-10-14:
Yes!! Orthogonal bit-width adjustment seems like the way to go. 🎊