calyxir/calyx #556: Register re-entrancy and pulse values

EclecticGriffin, 2021-06-23:
There are two current (related) issues that I am aware of with respect to registers in the interpreter. One is that they are not executed as often as required due to the current change based execution approach taken by group simulation. Fortunately, this is a relatively easy fix as we simply need to ensure that as long as the register's write_en is high it gets executed at every cycle, even when there's no change. In the limit, perhaps it makes sense to separate combinational and non-combinational cells and mandate that the interpreter re-run all non-combinational cells on every tick, regardless of whether any input values have changed.

The second issue is that when a register's write signal is held high, we expect the done signal to be low for a cycle and then remain high for N+1 cycles where N is the number of cycles that the write_en signal remains high. Currently the register will output a new pulsevalue that overwrites the existing one and will in-fact cause the register's done signal to remain low until cycle N and then high on N+1. Obviously this is incorrect, though the fix is not immediately obvious to me.

Perhaps these should be two separate issues.

sampsyo, 2021-06-24:
Sounds about right! For the first issue:

> In the limit, perhaps it makes sense to separate combinational and non-combinational cells and mandate that the interpreter re-run all non-combinational cells on every tick, regardless of whether any input values have changed.

Is it correct to assert that this would be "just an optimization," and it would also be correct to rerun _every_ component on _every_ cycle, even if there's no change? If so, maybe it's a good idea to approach it that way to avoid premature optimization—that is, start by removing the change sensitivity altogether and then, if it becomes a bottleneck, come back along later and add the distinction for speed alone.

EclecticGriffin, 2021-06-24:
Yes I believe that is correct! I think it makes sense to de-complexify the simulation logic as described since it gives us fewer failure points initially.

As for the latter, I think executing a primitive could be done more directly with the environment as context to avoid incorrect value trampling, which, if we're clever could free the interpreter from needing to extract the input port values on its own.

EclecticGriffin, 2021-06-24:
This also gives me a good idea about how to unify group & continuous interpretation w/o significant code duplication

PR #562 (2021-06-29): Simplify group simulation and handle repeated register execution
Closes #561.
Closes #556.

The major change here was switching the simulation logic to a simpler non-change-based approach. To accommodate this, updates to primitives have been made in a two-stage way so that their internal state does not get overridden in the event of superfluous execution. The core simulation loop has been refactored into a single function `interp_assigns` which can be used to drive, group interpretation and continuous only interpretation w/o copying logic anywhere.

Also removed the special printing for the si [... cut]
closed 2021-06-29 by EclecticGriffin