calyxir/calyx #557: What compiler passes should we ensure run before interpretation?

EclecticGriffin, 2021-06-23:
It seems pretty clear that we should run the basic papercut and validation passes (this is already done) but there may be merit is standardizing a set of the basic transformations that run before interpretation.

sampsyo, 2021-06-24:
One constraint that I think it would be cool to respect is that the interpreter _can_ run any Calyx program, before or after _any_ transformation. So from that perspective, it's great to run the papercut & validation passes because they are analysis only, but it may be better to not run any optimization or lowering passes by default. Instead, it can be the user's choice which "level of abstraction" they want to simulate at. Would that make sense?

rachitnigam, 2021-06-26:
Just to state the obvious, the tension here is writing a generic interpreter that can run every calyx program vs. writing a more specialized interpreter that can assume programs are in a specific state before running them.

For example, collapse-control prunes away unnecessary nestings of seq and par nodes. One obvious interpreter benefit of requiring these to be run is that when interpreting par nodes, you don’t have to fork the environment unnecessarily.

rachitnigam, 2021-08-02:
Do we need to do anything for this issue @EclecticGriffin @sampsyo?

EclecticGriffin, 2021-08-02:
Nothing really springs to mind, we can probably close for the time being

sampsyo, 2021-08-02:
Agreed!

closed 2021-08-02 by sampsyo