calyxir/calyx #560: Allow interpreter to consume .data (json) files

EclecticGriffin, 2021-06-23:
We should be able to initialize memory state from data files to allow evaluation of programs relying on input & output memories.

sampsyo, 2021-06-24:
Yes!! This would be super good, as would generating memory-only files that match. I just mentioned this in https://github.com/cucapra/calyx/pull/558#pullrequestreview-691867844, but IMO the way to do this is to lean on fud to do the conversion—not to have the interpreter itself responsible for reading & writing this format. For two reasons:

* There's already code in fud to wrangle this kind of conversion for the purpose of Verilator execution. So we can just reuse that to convert into & out of the interpreter-native format.
* The conversion is sensitive to data types, and the interpreter doesn't know about data types. That is, someone has to convert to fixed & floating point numbers—and it seems like the interpreter (being type-unaware) should not be responsible for that.

EclecticGriffin, 2021-07-01:
Organizing a todo list for myself on the main tasks for this issue:

- [ ] [Fud] Invoke interpreter from Fud
- [ ] [Fud] Use Fud's conversions for values going to verilog to make a JSON of bit strings for the memory values
- [ ] [Interpreter] Adjust CLI args to take in the json of memory initializations
- [ ] [Interpreter] Set up interpreter logic to deserialize bitstrings into the internal bitstring representation
- [ ] [Interpreter] Validate the input memories dimensions against the cell in the program
- [ ] [Interpreter] Inflate memory definitions based on input (for the moment assume that the memories live in the main component, this may need to be changed later to reflect that memories may exist for other components w/ name collisions).

PR #583 (2021-07-06): Fud interpreter integration
An attempt to close #560.

Major changes include:

***fud support for invoking the interpreter***
  try `fud e --to interpreter-out ...`

***fud support for passing data files to the interpreter***
  try `fud exec --to interpreter-out tests/while.futil -s interpreter.data tests/while.futil.data `

***Interpreter can now take and initialize memories based on the data marshaled from fud***

***memories are now a flat vec with row-major ordering indexing***

***NEW SERIALIZATION CRIMES***

Anyway, this still needs a bit of work, mainly in the writing tests department.  [... cut]
closed 2021-07-06 by EclecticGriffin