calyxir/calyx #561: Simplify the group interpreter

EclecticGriffin, 2021-06-24:
Splitting from #556. Currently the group interpreter is too clever for it's own good and approaches cells and assignments in a changed based way. Instead we should just run all assignments and execute all primitives during each loop of the simulation. This will likely be marginally more inefficient but less error prone and we can always re-optimize later.

EclecticGriffin, 2021-06-24:
I have a sketch of this (and #556 as a necessary sub-step) visible in https://github.com/cucapra/calyx/tree/simplify_group_sim which matches the runt tests!

Still needs some cleanup and work, but may subsume some of the stuff @yn224 has been working on. I refactored some of the shared logic into a function based on the insights from his work.

EclecticGriffin, 2021-06-24:
TIL rust won't even parse the tests if you aren't running them

rachitnigam, 2021-06-25:
You can use the command `cargo check --all --all-targets` to get around this.

PR #562 (2021-06-29): Simplify group simulation and handle repeated register execution
Closes #561.
Closes #556.

The major change here was switching the simulation logic to a simpler non-change-based approach. To accommodate this, updates to primitives have been made in a two-stage way so that their internal state does not get overridden in the event of superfluous execution. The core simulation loop has been refactored into a single function `interp_assigns` which can be used to drive, group interpretation and continuous only interpretation w/o copying logic anywhere.

Also removed the special printing for the si [... cut]
closed 2021-06-29 by EclecticGriffin