calyxir/calyx #607: Fixed point constants broken for larger numbers

sgpthomas, 2021-07-21:
The value of constants are passed as a verilog parameter. Vast, the underlying library we use uses `i32` for parameter values. We perform this casting 
[here](https://github.com/cucapra/calyx/blob/c0c74c79d75aec51f5c8804f5d9ede65bd793528/calyx/src/backend/verilog.rs#L251). The implication of this is that any fixed point number that is represented as a binary number with >32 bits will not work correctly (will be cast to 0 I think).

We should
1) throw an error if there is a number we can't represent instead of silently just using some random number
2) update vast to support larger numbers

sampsyo, 2021-07-21:
Nice work root-causing it! Out of curiosity, do you think `i64` is enough, or do we have to go with [bitvec](https://docs.rs/bitvec/0.22.3/bitvec/) or [num_bigint](https://docs.rs/num-bigint/0.4.0/num_bigint/) something else arbitrary-precision?

rachitnigam, 2021-07-30:
Also pinging @vegaluisjose to this thread. It seems it might be best to fix it int vast itself?

vegaluisjose, 2021-07-30:
Hey guys, literals without width in Verilog say `35` are treated as 32-bit integers reason why we have `Expr::new_int()`, however, for larger numbers I would suggest using any of these three:

```
Expr::new_ulit_dec(…)
Expr::new_ulit_hex(…)
Expr::new_ulit_bin(…)
```

The arguments for these constructors are `(width, value)` and value is a string, so use it carefully..you can use `format!(“{:X}”, yourvalue)` to generate those for example.

I hope it helps!

sampsyo, 2021-07-30:
Neato; it does seem like the thing to do is to emit hex or decimal literals or whatever. Thanks, @vegaluisjose!

rachitnigam, 2021-08-01:
@sgpthomas do you want to take a shot at fixing this? I don't have the test file to reproduce the problem you were running into.

rachitnigam, 2021-09-12:
@sgpthomas can you add a file to reproduce the bug? I can fix it but need a regression test.

PR #954 (2022-03-11): correctly emit large constants
Fixes #607
closed 2022-03-11 by rachitnigam