calyxir/calyx #616: Questionable effects of applying different passes

yn224, 2021-07-30:
While thinking of issue #574, I came across some transitions in the code that was questionable. An output using the tool addressed in PR #615 is shown below.
```
======================================pass_seq======================================
0-well-formed.futil

1-papercut.futil

10-dead-cell-removal.futil
121:      fsm.in = fsm.out == 4'd2 & cond_if[done] ? 4'd3;
147:      fsm.in = fsm.out == 4'd2 & !cs_wh.out ? 4'd11;

11-go-insertion.futil
121:      fsm.in = fsm.out == 4'd2 & cond_if[done] & tdcc[go] ? 4'd3;
147:      fsm.in = fsm.out == 4'd2 & !cs_wh.out & tdcc[go] ? 4'd11;

12-component-interface-inserter.futil
121:      fsm.in = fsm.out == 4'd2 & cond_if[done] & tdcc[go] ? 4'd3;
147:      fsm.in = fsm.out == 4'd2 & !cs_wh.out & tdcc[go] ? 4'd11;

13-hole-inliner.futil
81:    fsm.in = fsm.out == 4'd2 & 1'b1 & go ? 4'd3;
107:    fsm.in = fsm.out == 4'd2 & !cs_wh.out & go ? 4'd11;

14-clk-insertion.futil
81:    fsm.in = fsm.out == 4'd2 & 1'b1 & go ? 4'd3;
107:    fsm.in = fsm.out == 4'd2 & !cs_wh.out & go ? 4'd11;

15-reset-insertion.futil
81:    fsm.in = fsm.out == 4'd2 & 1'b1 & go ? 4'd3;
107:    fsm.in = fsm.out == 4'd2 & !cs_wh.out & go ? 4'd11;

16-merge-assign.futil
54:    fsm.in = fsm.out == 4'd2 & !cs_wh.out & go ? 4'd11;
57:    fsm.in = fsm.out == 4'd2 & 1'b1 & go ? 4'd3;

2-guard-canonical.futil

3-infer-static-timing.futil

4-collapse-control.futil

5-resource-sharing.futil

6-minimize-regs.futil

7-compile-invoke.futil

8-compile-empty.futil

9-top-down-cc.futil
124:      fsm.in = fsm.out == 4'd2 & cond_if[done] ? 4'd3;
150:      fsm.in = fsm.out == 4'd2 & !cs_wh.out ? 4'd11;
```
When `TopDownCompileControl` pass gets applied, we start seeing `fsm` getting introduced and the lines that cause conflict for `iter_mult.futil`. Would the first line also include `cs_wh.out` as part of its guard to make the two statements mutually exclusive?

Also, after the `Inliner` pass gets applied, the signal `cond_if[done]` seems like it transforms into `1'd1`. Why is that?

Further investigations of these points would help me address issues related to `iter_mult` test case.

sampsyo, 2021-07-30:
Thanks, @yn224! For self-contained completeness, can you please remind us how to reproduce the problem? That is, giving us the `fud` command to run (on which file, possibly a reduced version thereof) to observe the conflicting assignments could be very helpful.

yn224, 2021-07-31:
For the error, we can run 
`fud e ~/interp/tests/control/iteration/iter_mult.futil -s futil.flags '-p all -d static-timing' --to futil-lowered`
to produce a panic:
`[ fsm.in = fsm.out == 4'd2 & !cs_wh.out & go ? 4'd11; ] and [ fsm.in = fsm.out == 4'd2 & go ? 4'd3; ]`

Or replace the test file with a simplified version
```
import "primitives/core.futil";

component main() -> () {
  cells {
    zero = std_const(1, 0);
  }

  wires {
    group cond_while {
      cond_while[done] = 1'b1;
    }

    group cond_if {
      cond_if[done] = 1'b1;
    }
  }

  control {
    while zero.out with cond_while {
        if zero.out with cond_if {
          seq {}
        } else {
          seq {}
        }
    }
  }
}
```
which will also produce a conflicting assignment panic
`Between assignments [ fsm.in = fsm.out == 3'd1 & go ? 3'd2; ] and [ fsm.in = fsm.out == 3'd1 & !cs_wh.out & go ? 3'd5; ]`

rachitnigam, 2021-07-31:
My guess is that again, the assignment `fsm.in = fsm.out == 4'd2 & cond_if[done] ? 4'd3;` is too aggressive and should probably be using the `cs_wh` in some capacity before performing the transition. I still have to figure out where exactly the problem code originates.

sampsyo, 2021-07-31:
This makes sense. I wonder if the fix for this specific case might be to just include `& cs_wh` for FSM transitions like this, but it requires some experimentation.

I've changed the title of the issue to be specific about what we think the problem is.

rachitnigam, 2021-08-02:
Changes in #620 should help with this. @yn224, can you try running the same example on the #620 branch (if it's not merged yet).

yn224, 2021-08-02:
@rachitnigam The PR you mentioned seems to resolve this issue.

rachitnigam, 2021-08-23:
#620 fixes this issue

closed 2021-08-23 by rachitnigam