calyxir/calyx #639: Generate size efficient FSMs in `top-down-cc`

rachitnigam, 2021-08-31:
While fixing FSM generation in #635, I sketched out what the size-optimal FSMs (in the number of states) may look like. As far as I can tell, generating these FSMs require more dramatic changes to the FSM generation process than that I would like to do right now so I'm leaving this as a future issue.

```
seq {
  cond0;
  while comb_reg.out {
    seq {
      incr;
      cond0;
    }
  }
}
```

States:

0: enable cond0
1: enable incr
2: enable cond0

(0, 1): cond[done] & comb_reg.out
(0, 3): cond[done] & !comb_reg.out
(1, 2): incr[done]
(2, 1): cond0[done] & comb_reg.out

```
cond0;
if lt.out {
  true;
} else {
  false;
}
if lt.out {
  true;
} else {
  false;
}
```

0: enable cond0
1: enable true
2: enable false
3: enable true
4: enable false

(0, 1): cond0[done] & lt.out
(0, 2): cond0[done] & !lt.out
(1, 3): true[done] & lt.out
(1, 4): true[done] & !lt.out
(2, 3): false[done] & lt.out
(2, 4): false[done] & !lt.out

```
cond0;
if lt0.out {
  if lt1.out {
    one;
  } else {
    two;
  }
} else {
  three;
}
```

0: enable cond0
1: enable one
2: enable two
3: enable three

(0, 1): cond0[done] & lt0.out & lt1.out
(0, 2): cond0[done] & lt0.out & !lt.out
(0, 3): cond0[done] & !lt0.out

sampsyo, 2021-08-31:
Sounds intuitively interesting… someday when you have a chance, it would be cool to hear more about what you mean by "size-optimal FSMs" (i.e., what optimizing the number of states would look like).

rachitnigam, 2021-09-02:
"size-optimal FSMs" is a terrible name given that I don't know if it's optimal in any way at all. The thing that made me think that this representation of FSMs is more concise is because, in example two, the branches of the conditional can immediately jump into the branches of the other conditional without requiring an extra transition.

Anyways, I ended up implemented this in #635 so I can show what the generated FSMs look like.

closed 2021-09-07 by rachitnigam