calyxir/calyx #643: Deprecate `std_const`.

cgyurgyik, 2021-09-01:
I think this has come up in discussion before, but it seems just using the SystemVerilog representation e.g. `1'b1` is fine.

Pros: 
- One representation for constants. 
Cons: 
- Lots of refactoring.
- ???

rachitnigam, 2021-09-02:
The big problem is that the Dahlia frontend's compilation process can only refer to ports on defined cells. Constants are magical in that they are represented by a hidden cell on the component and not explicitly defined. Is there any benefit to deprecating constants?

cgyurgyik, 2021-09-05:
Gotcha, didn't think about that. Yeah, one representation = simpler? :).

rachitnigam, 2021-09-06:
Yeah, maybe there is a way to make the frontend treat `std_const` declarations specially somehow and intern the constants.

rachitnigam, 2021-09-10:
The dumb approach of just interning a constant in the `add_cell` method of `from_ast` does not work because the assignments will still have reference to the name of the defined cell.

A better approach may be to first define the cells and then deduplicate them once the component has been created.

rachitnigam, 2021-09-12:
Hm, I'm actually not so sure that this a good idea. We don't have a first-class representation for fixed-point and signed constants and might eventually add a `std_sconst` and `std_fp_const` primitive right?

cgyurgyik, 2021-09-12:
There was an incorrect `std_fp_const` at one point. However, I removed it since (1) it was incorrect, and (2) the representation of a constant is always just an array of bits. We depend on the primitive (e.g. `std_fp_mult` versus `std_mult`) to correctly interpret these.

rachitnigam, 2021-09-12:
The point of a `std_fp_const` primitive would be that the compiler/programs can interpret the bits using the representation defined in the cell definition:
```
fp = std_fp_const(32, 8, 1000);
```

Do you think this would be useful or no?

cgyurgyik, 2021-09-12:
Usefulness: Frontends don't need to do the math to convert `3'd0.5` into the correct unsigned integer. If we do that, wouldn't we want to do it for all types of constants? `fp, signed_fp, bit, signed_bit`. 

Additional questions: At that point, are we turning this into a type system? Would we add verifiers to ensure that `signed_fp` is only being fed into `signed_fp` primitives?

sampsyo, 2021-09-12:
Popping in here to say: I'm concerned that adding `std_fp_const` would not solve the problem of telling programmers/tools how to interpret bits. For example, as soon as you store a value into a memory, we'd lose the type information. A "proper" solution to this kind of debuggability information would be the optional "overlay" type system idea we've jotted in our [idea list](https://github.com/cucapra/calyx/projects/2). To put it differently, IMO, we'd want to distinguish the types on ports (using attributes), not cells.

And as @cgyurgyik says, we'd want such information to be able to cover more types than just integer and FP…

rachitnigam, 2021-09-13:
Fair points.

However, an overlay type system (that checks for constants) will not work if the cells are not represented in the IR. We can choose to deprecate std_const but make it an internal IR node in the compiler if we still want an explicit representation.

rachitnigam, 2021-09-24:
During the meeting today we decided that there is value in having `std_const` available to frontends as a normal cell and that the IR can probably normalize them at a future point if needed. Closing this for now since we've decided to not deprecate `std_const`.

closed 2021-09-24 by rachitnigam