calyxir/calyx #646: [CIRCT] Incorrect port ordering.

cgyurgyik, 2021-09-01:
https://github.com/cucapra/calyx/blob/2fa34968ec1118599e4c4847b8ac757f43a116a6/src/backend/circt.rs#L212-L225

I think the port name and width ordering may not always align with the Calyx dialect.

E.g. the memory example in #645.

rachitnigam, 2021-09-02:
@cgyurgyik do you have a sense of why? The port ordering in the generated Calyx should be the same as the input definition. If not, we need to fix the frontend/backend.

cgyurgyik, 2021-09-02:
Not sure I understand the question. To be more clear, I meant that, for example,

```
let all_ports = cell 
     .ports() 
     .iter() 
     .map(|p| format!("%{}.{}", name, p.borrow().name)) 
     .collect::<Vec<_>>() 
     .join(", "); 
```

may have a different ordering of port names / bit widths than what a memory has in the Calyx dialect:

```
      // `m1 = std_mem_d1(32, 1, 1)`
      %m1.addr0, %m1.write_data, %m1.write_en, %m1.read_data, %m1.done = calyx.memory "m1"<[1] x 32> [1] : i1, i32, i1, i32, i1
```

For instances of components, I imagine we should always be fine. Does this make sense?

cgyurgyik, 2021-09-02:
Another option would be to change the primitives in the Calyx Dialect to align with the native compiler's port ordering. For example, I chose the ordering for RegisterOp arbitrarily. Not sure if @mikeurbach did the same for MemoryOp [here](https://github.com/llvm/circt/blob/0066f6dc79c8d558d4168922ee370b786b8dcf9f/lib/Dialect/Calyx/CalyxOps.cpp#L749-L789).

mikeurbach, 2021-09-02:
For MemoryOp in CIRCT, I decided to just add one MemoryOp with support for an arbitrary number of dimensions, rather than different ops for different numbers of dimensions. It seemed easier to make the first N results (for N == number of dimensions) be the address ports. I wasn't thinking about aligning with the Rust compiler, so if it makes sense to adjust the CIRCT side we definitely can.

cgyurgyik, 2021-09-02:
Ok Rachit if you open up an issue with correct port ordering in the CIRCT repository I'll do the refactoring when I have a chance.

rachitnigam, 2021-09-03:
Done: https://github.com/llvm/circt/issues/1706

cgyurgyik, 2021-09-04:
Thanks! Addressed [here](https://github.com/llvm/circt/issues/1706#issuecomment-912673255).

closed 2021-09-04 by cgyurgyik