calyxir/calyx #649: Get unrolled Polybench benchmarks working with interpreter

rachitnigam, 2021-09-03:


EclecticGriffin, 2021-09-09:
Seems likely that there's a bug in the state merging function for the stack environments, I'll update when I have a better sense of it's precise location & nature

EclecticGriffin, 2021-09-10:
Okay I have determined the bug which is that `eval_empty` assumes that it's being run on a structural component (since that's sometimes what an empty group means) and it waits for the component to finish. This is obviously not the case when we're executing the empty branch of an if statement (the false branch) so it spins forever.

The solution is to fix the top-level check to see if the component is structural to appropriately dispatch `interp_cont` and `eval_empty` should just be a pass-through.

I should note that with this fix the unrolled benchmarks crash because multiple arms of a par write the same value to the same port

PR #663 (2021-09-10): Patch inifinite unroll bug
Closes #649.

Fixes a silly bug causing incorrect behavior on empty groups when running through the normal interpreter pass.

The unrolled ones now crash due to multiple-par writes, as anticipated.

~Note: may wish to use the `control_is_empty` function from the stepper utils rather than only matching a top level empty.~ done
closed 2021-09-10 by rachitnigam