calyxir/calyx #669: Example external `fud` stage

mortbopet, 2021-09-14:
While there exists some documentation for adding external stages to fud (https://capra.cs.cornell.edu/docs/calyx/fud/index.html#adding-external-stages), there is (as far as i can tell) no actual examples available for out-of-tree stages. I suspect that with an example to reference, I could have saved a good 30 minutes on trying to set up an external stage.

rachitnigam, 2021-09-14:
@sgpthomas if you have an external stage lying somewhere, can you add a link to it? We can try to move something like mrxl to be an external stage so that we eventually have an in-tree example.

sgpthomas, 2021-09-14:
Here's an example: https://github.com/sgpthomas/mandlebrot-fpga/blob/main/stage.py. Albeit this one is a little messy. I think moving mrxl to an external stage is probably a good idea. And I suppose we can now link to the circt one.

PR #674 (2021-09-15): [fud] External stages revamp
External stages are now stored in the [stages] array of the configuration and
support `-s` flag natively.

- Made `mrxl` an external stage and added doc instructions on how to use it
- Added doc instructions on defining new stages. (Fixes #669)
- `-s` works as expected with external stages. (Fixes #670)
- `fud check` will check the binary mentioned in the `exec` field of an external stage.
commit 27750460 (2021-09-15): [fud] External stages revamp (#674)

External stages are now stored in the [stages] array of the configuration and
support `-s` flag natively.

- Made `mrxl` an external stage and added doc instructions on how to use it
- Added doc instructions on defining new stages. (Fixes #669)
- `-s` works as expected with external stages. (Fixes #670)
- `fud check` will check the binary mentioned in the `exec` field of an external stage.
closed 2021-09-15 by rachitnigam