calyxir/calyx #670: Overriding configuration key-value pairs only works for built-in stages

mortbopet, 2021-09-14:
Having an external stage
```
[external-stages.circt]
location = ...
exec = ...
toplevel = "main"
```
it is currently not possible to manipulate the `toplevel` through the `-s` parameter of `fud exec`. Only built-in stages (i.e. stages with a `[stage.${stagename}]` header in the config file) can have their key/value pairs overridden.

rachitnigam, 2021-09-14:
Thanks for catching this @mortbopet! If possible, could you provide a gist of the external stage you were trying to add so that we can debug it directly?

mortbopet, 2021-09-14:
Sure! I'm working on https://github.com/circt-hls/circt-hls/blob/main/stages/CIRCT/stage.py where I'd like to run with different values for the `circt.toplevel` variable:
```
fud exec "Polygeist/mlir-clang/Test/aff.c"  \
  --from c                                  \
  --to mlir-calyx                           \
  -s circt.toplevel "kernel_deriche"`
```

however, I'm having to resolve to using a configuration like the following:
```
[stages.circt]
toplevel = ""

[external-stages.circt]
location = "/home/morten/Work/circt-hls/stages/CIRCT/stage.py"
bin_dir = "/home/morten/Work/circt-hls/../circt/build/bin/"
llvm_bin_dir = "/home/morten/Work/circt-hls/../circt/llvm/build/bin"
```
with `toplevel` being listed under a `[stage.circt]` header, instead of `[external-stages.circt]`. As far as i can tell, the `-s` argument parsing assumes that the provided configuration key is listed as a `stage` and does not consider `external-stage`s.

rachitnigam, 2021-09-14:
Oh, fantastic! This looks like a super cool way to use `fud`. I'm going to tag @EclecticGriffin & @cgyurgyik if they get the time to look at the bug before I do.

mortbopet, 2021-09-14:
So far I'm really liking the driver! ... I'm only wondering about why it's nested within Calyx and not a separate project - something like `fud` could be really nice for a general lightweight/standalone pass manager infrastructure.

rachitnigam, 2021-09-14:
Largely because it started off as a way to run Calyx-related tools and we don't have any external users yet 😅. I think it makes a lot of sense to move it into a separate project in the long-term because it supports a lot of tools now. We'd need to document it better and generalize the code but if this sounds like something you'd be interested in helping with, we can create an issue to get started!

PR #674 (2021-09-15): [fud] External stages revamp
External stages are now stored in the [stages] array of the configuration and
support `-s` flag natively.

- Made `mrxl` an external stage and added doc instructions on how to use it
- Added doc instructions on defining new stages. (Fixes #669)
- `-s` works as expected with external stages. (Fixes #670)
- `fud check` will check the binary mentioned in the `exec` field of an external stage.
commit 27750460 (2021-09-15): [fud] External stages revamp (#674)

External stages are now stored in the [stages] array of the configuration and
support `-s` flag natively.

- Made `mrxl` an external stage and added doc instructions on how to use it
- Added doc instructions on defining new stages. (Fixes #669)
- `-s` works as expected with external stages. (Fixes #670)
- `fud check` will check the binary mentioned in the `exec` field of an external stage.
closed 2021-09-15 by rachitnigam