calyxir/calyx #672: Lowering combinational groups and components

mortbopet, 2021-09-14:
The recent move to combinational groups indeed had some breaking changes for https://github.com/llvm/circt/pull/1630.

Before, something like the following was a valid calyx program:
```
component main(in0: 24, in1: 24, @clk clk: 1, @reset reset: 1, @go go: 1) -> (out0: 8, @done done: 1) {
  cells {
    std_slice_0 = std_slice(32, 8);
    std_add_0 = std_add(32);
    std_pad_1 = std_pad(24, 32);
    std_pad_0 = std_pad(24, 32);
  }
  wires {
    group bb0_0 {
      std_pad_0.in = in1;
      bb0_0[done] = 1'd1;
    }
    group bb0_1 {
      std_pad_1.in = in0;
      bb0_1[done] = 1'd1;
    }
    group bb0_2 {
      std_add_0.left = std_pad_0.out;
      std_add_0.right = std_pad_1.out;
      bb0_2[done] = 1'd1;
    }
    group bb0_3 {
      std_slice_0.in = std_add_0.out;
      bb0_3[done] = 1'd1;
    }
    out0 = std_slice_0.out;
  }
  control {
    seq {
      bb0_0;
      bb0_1;
      bb0_2;
      bb0_3;
    }
  }
}
```
With the restriction of combinational groups (as far as i understand ) **not** being allowed inside a schedule, i.e. only for if/while `with` groups.

After the introduction of combinational groups, is it now expected of frontends to:
- Group together all combinational logic into the `with` group of the nearest following if/while group
- and if no if/while group is present, combinational statements should be nested directly under the `wires` operation?

rachitnigam, 2021-09-14:
Yup, the addition of combinational groups is to precisely specify the semantics of programs like these. Calyx's group isolation property ensures that the *only* way to observe changes outside a group is using stateful elements. This means that combinational groups cannot affect anything in normal Calyx components because they cannot modify state.

For example, in this program, I'm pretty sure the intention is to perform the computation:
```
out0 = ((in0 as i32) + (in1 as i32) as i8)
```

The way the [Dahlia frontend](https://github.com/cucapra/dahlia/) will lower this computation:
```
let in0: ubit<24>;
let in1: ubit<24>;
let out0: ubit<8> = ((in0 as ubit<32>) + (in1 as ubit<32>) as ubit<8>);
```
by generating:
```
// git.status = clean, build.date = Fri Sep 10 13:50:44 EDT 2021, git.hash = d68a833
import "primitives/std.lib";
component main() -> () {
  cells {
    add0 = std_add(32);
    in0_0 = std_reg(24);
    in1_0 = std_reg(24);
    out0_0 = std_reg(8);
    pad0 = std_pad(24,32);
    pad1 = std_pad(24,32);
    slice0 = std_slice(32,8);
  }
  wires {
    group let0<"static"=1> {
      out0_0.in = slice0.out;
      out0_0.write_en = 1'd1;
      let0[done] = out0_0.done;
      slice0.in = add0.out;
      add0.left = pad0.out;
      add0.right = pad1.out;
      pad0.in = in0_0.out;
      pad1.in = in1_0.out;
    }
  }
  control {
    let0;
  }
}
```

In general, Dahlia might provide useful examples for how to lower these programs.

rachitnigam, 2021-09-14:
@mortbopet if you feel this is answered, feel free to close.

mortbopet, 2021-09-15:
Yep, i think i got what i needed, thank you!

closed 2021-09-15 by mortbopet