calyxir/calyx #676: Interpreter internal reorganization and debugger hierarchical printing

EclecticGriffin, 2021-09-15:
The debugger needs the ability to inspect values from within subcomponents, however, due to issues with the current lifetime setup this is far more onerous than it ought to be.

Fixing this will require some rearchitecting of the current interpreter to reduce or otherwise eliminate the lifetime issues.

rachitnigam, 2021-09-24:
@EclecticGriffin would you say that we have actionable steps to make progress on this? If so, can you remove the discussion tag?

EclecticGriffin, 2021-09-24:
I think there's still some open questions about how to navigate the state representation. The tree representation encapsulates over some of the multiplicities that can arise from nested `par`s and I'm not entirely sure how to do the appropriate representation in a flatter context.

One option might be to take a hybrid approach and retain a tree like structure but with backpointers to the control RRC. I'm not entirely sure if that would be any better though.

Another potential decision would be to revisit the "multi-verse" model of parallel computation and adjust the environment accordingly

EclecticGriffin, 2021-09-30:
Just to elaborate slightly on what was talked about in the meeting today with regard to adjustments to the way the interpreter deals with the control structure. The core issue is that we would like computational pieces of the interpreter to be able to hold a reference to the control structure without requiring special lifetimes which means we want a reference which is known to always be good as long as it's available. There are a few possible choices.

1. Completely flatten the interpreter structure and always work from the top.

This is probably not a great choice as it would necessitate massive rewrites and would require us to entirely re-think what the interpreter state is.

2. Have the interpreter blocks hold a raw pointer to the section of the control they are working on and use the top-level RC to walk the control tree until a matching control block is located. 

This would fix the lifetime issues but does still require some walking of the control structure which imposes some complexity and cost.

3. Do the above but dereference the raw pointers.

This is unsafe and also probably not a wise approach to take. Though it is potentially feasible if done carefully it is likely not worth the trouble.

4. RC-ify the control structure within the interpreter.

This would neatly solve the lifetime issues, but it does require having a variant of the IR structures which are used within the interpreter and in general means there is some amount of code and definition duplication, though it is hopefully not to odious.

sampsyo, 2021-09-30:
Awesome; thanks for jotting down these notes, @EclecticGriffin. I do generally like the idea of RC-ifying the control statements so they can be shared… even if there's some duplication, it makes me wonder how bad it could possibly be! Option 2, with the raw pointers and walking down the structure to find things, strikes me as a possible runner-up because the complexity seems like it could be isolated from the rest of the interpreter implementation.

closed 2021-10-08 by EclecticGriffin